bambulab/BambuStudio #967: SD Card File Storage Improvements

TheHogarthWorkshop, 2022-12-28:
- Icon on screen should show layer height of print as currently only shows weight and ETA.
- SD Card Files should be viewable on Bambu Studio APP with browser like function 'Create Folders etc'
- I send a lot of files to SD over the day and loose track easily needs timestamps at least and/or displayed on touch screen as oldest/nonprinted first - printed should go to a printed or completed folder.

Ideally, having dates folders option would work, so when sending the file to printer, it does the folder structure based on Year/Month/Day or at least have the option to filter in a similar fashion. 

Cheers
Phil

KD7LRJ, 2022-12-28:
1. Layer height can be variable. For me, time and weight are much more important for the limited space available (especially the P1P).
2. I agree that file management could be improved. Don't really need this on the printer if BS and Handy could start prints from the files on the SD card.
3. Yes, sorting the files would be good. I want mine sorted by newest files first most of the time. If folders become available, I may want them sorted alphabetically. Like you mentioned, options would be great!

TheHogarthWorkshop, 2022-12-28:
1. Layer height if variable could just say 'variable' but yeah, the space is an issue there but that could be sorted with making it a list format instead of thumbnails, I want to see more data, and at a glance. Maybe options/customise

CCS86, 2023-01-17:
Agreed that we should have file management options in BS and BH for SD and internal storage. 

I would also add, we should be able to send files to the SD card from a different network. In my case, our ethernet and WIFI LANs are on different subnets, so I can't even send files to the SD card. Seems like a false constraint.