bambulab/BambuStudio #969: Request for adding Frimware Retract

Eifel-Joe, 2022-12-28:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I'd like to have the possibility to choose between slicer retract and firmware retract (G10/G11).

**Describe alternatives you've considered**
For now i use slicer retract.

**Additional context**
I only have klipperized printers. I love the possibility to change retract settings during print in klipper suing Klippersreen. It would be nice to add this solution in your fork of BambuStudio.

SaltWei, 2023-08-18:
This is not printer issue @bambu123 . But feature request for Studio of 3-rd printer.
@XunZhangBambu Please handle it.

XunZhangBambu, 2023-11-14:
In 1.8 beta ,we add this feature

closed 2023-12-21 by QingZhangBambu