bambulab/BambuStudio #971: Additional Extruder head movement after filament change.

Praxis459, 2022-12-28:
**Describe the bug**
After a filament color change the extruder head will move from the filament waste area to what looks like the home location very quickly, then move back to the filament waste area and then over to the prime tower.  That first movement to the home point has caused collisions with the print job that is currently running.

**3mf File for This Bug**
https://drive.google.com/file/d/1E83pJDzCoG_S6jSOKzgGEwac3Stlp4Ni/view?usp=share_link

**To Reproduce**
Steps to reproduce the behavior:
1. Slice current file.
2. Send sliced file to printer
3. Wait for printer to print model movement error will eventually show up, may or may not cause collision with model.

**Expected behavior**
The print head should move from the filament waste area directly to the prime tower area and not move to the home location.

**Screenshots**
Here is a video showing the movement 
https://user-images.githubusercontent.com/19175365/209854690-b4a509ac-bda0-4d89-92bc-c2a708acbfb2.mp4

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.4.0.17

SaltWei, 2022-12-29:
This is a known issue caused by "avoid crossing wall".

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei