bambulab/BambuStudio #972: Copy results in parts beeing placed in "Outside"

zenturacp, 2022-12-28:
**Describe the bug**
I have a specific part - when i copy it the first part is always on Plate 1 - but subsequent parts is always in the Outside group, im wondering what "Outside" is

**3mf File for This Bug**
[Cover.zip](https://github.com/bambulab/BambuStudio/files/10316538/Cover.zip)

**To Reproduce**
Steps to reproduce the behavior:
Add the file to the build area - copy / paste it - subsequent parts goes to the Outside folder and the slicing seems wrong, the brim disapears sometimes

**Expected behavior**
All the parts to be on plate 1 - there is more than enough space for this

**Screenshots**
[image]

**Desktop (please complete the following information):**
Windows 11
Bambu Studio 1.4.0.18

SaltWei, 2022-12-29:
Thank you very much for reporting issue to us to make software improved!
Is it always reproduced on your PC?
We can't reproduce this issue and guess maybe it's occasional.

zenturacp, 2022-12-29:
> Thank you very much for reporting issue to us to make software improved!
> 
> Is it always reproduced on your PC?
> 
> We can't reproduce this issue and guess maybe it's occasional.

It does it every time I copy it .. this object - i can share a video

lanewei120, 2022-12-30:
> > Thank you very much for reporting issue to us to make software improved!
> > Is it always reproduced on your PC?
> > We can't reproduce this issue and guess maybe it's occasional.
> 
> It does it every time I copy it .. this object - i can share a video

I also can not reproduce it
[image]

after you load this stl, have you done some other steps?

could you give a full video from the beginning?

zenturacp, 2022-12-30:
> > > Thank you very much for reporting issue to us to make software improved!
> > > Is it always reproduced on your PC?
> > > We can't reproduce this issue and guess maybe it's occasional.
> > 
> > 
> > It does it every time I copy it .. this object - i can share a video
> > @zenturacp
> > I also can not reproduce it
> > [image]
> 
> after you load this stl, have you done some other steps?
> 
> could you give a full video from the beginning?

https://youtu.be/idN-hQd1bOg

Here is it - i forgot i also changed orientation of the file..

zenturacp, 2022-12-30:
> Thank you very much for reporting issue to us to make software improved!
> 
> Is it always reproduced on your PC?
> 
> We can't reproduce this issue and guess maybe it's occasional.

Look at the YouTube link i added to the issue..

lanewei120, 2022-12-31:
@zenturacp 
thanks for the detailed video, I can reproduce it on 1.4.0.18
and we found this issue has been fixed under 1.4.1

could you have a try using V1.4.1
https://github.com/bambulab/BambuStudio/releases/tag/v01.04.01.52

zenturacp, 2023-01-01:
> @zenturacp thanks for the detailed video, I can reproduce it on 1.4.0.18 and we found this issue has been fixed under 1.4.1
> 
> could you have a try using V1.4.1 https://github.com/bambulab/BambuStudio/releases/tag/v01.04.01.52

All is perfect on that release!! :-)

This is the reason i recommend this printer!! REALLY good support.. thx thx.. now i just need one my self! :-D

closed 2023-01-01 by zenturacp