bambulab/BambuStudio #973: [Feature Request] Allow objects to move below plate plane and only print the parts that are above plate

BambuLabDiscord, 2022-12-28:
Currently BS doesn't allow any part of an object to move below the plate plane. Allowing objects to go below the plate, like in Cura, will save time in situations such as testing out overhang, or reprinting part of a big model that previously failed at a certain height. In these cases, the lower part of the model doesn't need to be printed.

SaltWei, 2022-12-29:
Same with https://github.com/bambulab/BambuStudio/issues/585

TheHogarthWorkshop, 2022-12-29:
This is actually supported in Superslicer,

wirlybird, 2022-12-31:
This would be a good feature but it should also allow the part to be lifted from the build plate to create a gap.
This allows supports to go under a part when the second or third layer are too close to the build plate for supports.

msteele999, 2023-01-12:
I would like this as well. I have some models that need to be dropped by 1mm below the plate to make their bases flush on the first layer

SaltWei, 2023-04-18:
Has been included in v1.6 public beta

closed 2023-04-18 by SaltWei