bambulab/BambuStudio #975: [and can you please add raft air gap to your bambu slicer please and thank you] raft air gap

BambuLabDiscord, 2022-12-29:
Separation Distance – This is an important setting that defines the height of the air gap between the raft and your part. Typically, having a gap of at least 0.1mm will help keep the part connected to the raft, while still allowing for easy separation once the print is complete.

The Raft Air Gap setting is simply how large the gap is between the raft and the model itself. The larger this gap is, the easier it is to remove. It basically

SaltWei, 2022-12-30:
Accept! We have marked this in feature wish list

wirlybird, 2022-12-31:
I just ask for this also!

c2h5oh, 2023-01-02:
I would have liked that feature yesterday - I've spent half an hour doing cleanup

SaltWei, 2023-01-03:
Will be added in next release.

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei