bambulab/BambuStudio #976: Slicing Negative Parts 2 Bugs found

Benderrodrigu3z, 2022-12-29:
This is from https://www.printables.com/model/321749-bambu-lab-ams-hub-remove-button/comments

When you hit slice it shows a print tower because the negative part was once a physical part and has a filament color roll attached to it but its a negative part...

Worse yet if you want to abs the parts or any other material you have to convert the negative part(s) to a part then change the new part to abs then back to a negative part

MacOS 10.15.7
there are other comments on the page describing similar issues

[AMS_Hub_PTFE_Button_V2.3mf.zip](https://github.com/bambulab/BambuStudio/files/10320535/AMS_Hub_PTFE_Button_V2.3mf.zip)

SaltWei, 2022-12-30:
> AMS_Hub_PTFE_Button_V2.3mf.zip

For this model, the wipe tower is not caused by negative part. It‘s caused by different filament for support.
[image]

Benderrodrigu3z, 2022-12-30:
you are correct, my bad, didn't realize that one of my profiles still had that selected thanks, I'll tell the uploader

Benderrodrigu3z, 2022-12-30:
however the negative part is still a thing on going to the other color, I have tested and confirmed it

Benderrodrigu3z, 2022-12-30:
[image]
[image]

The Slicer removes the white only after slicing it but it still shows as a tower on the prepare screen, this has confused some users on the Discord and in Printables

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei