bambulab/BambuStudio #978: Bridging fan speed lagging

leofaoro, 2022-12-29:
**Describe the bug**
in bridging moves, the fan is going to bridge speed way after the bridge has already started, resulting in saggy bridging. using PETG. Bridging fan speed is shown as expected in preview (see screenshot)

**3mf File for This Bug**
[BridgingTest%20v1.3mf.zip](https://github.com/bambulab/BambuStudio/files/10321858/BridgingTest.20v1.3mf.zip)

**To Reproduce**
slice and print as usual

**Expected behavior**
better bridge quality

**Screenshots**
[image]

**Desktop (please complete the following information):**
MacOS Ventura

SaltWei, 2022-12-30:
This is a feature request to control fan in advance.

SaltWei, 2022-12-30:
By the way, this bridge is too long, please enable thick bridge and support to try again.

leofaoro, 2022-12-30:
> This is a feature request to control fan in advance.

Thanks for your response. No this is because after the nozzle is already printing the bridge, it takes seconds until the fan goes up to the bridge % (90% in my case)

> By the way, this bridge is too long, please enable thick bridge and support to try again.

This model is a calibration print, to allow me to optimize bridging. In normal use I'd use supports, of course. thanks

psiberfunk, 2023-01-11:
Depending on how you look at this @SaltWei this is either a bug or a feature.  "Fan prespin" as a general slicer feature for more accurate fan timing could be considered a feature... 

However, given the integrated nature of the Bambu X1/P1P systems you could consider this a bug since you know **exactly** how long it takes your fan to spin up/down and ramp, you arguably should have pre-baked values for the X1/P1P put into your slicer for your own printers that makes the printer behave as intended (e.g. have the right fan speed at the right time).  Since it's NOT acting as intended (as shown in the preview), and you know _a priori_ the behavior of the cooling fan (since you make a vertically integrated printer and slicer product offering), it is also reasonably classified as a bug.

In either case, this should be not too hard to implement and test, and adds real value.  It's also already in SuperSlicer :)