bambulab/BambuStudio #979: Wrong extrusion created

Snicket8247, 2022-12-29:
**Describe the bug**
When slicing the attached file at layer 89 an extrusion across the object is created instead of a travel move(?).
This only happens at 0.28mm layer hight IF the initial layer hight is set to 0.28mm as well.

**3mf File for This Bug**
[Kreidenhalter Bambu Studio Bug.zip](https://github.com/bambulab/BambuStudio/files/10321980/Kreidenhalter.Bambu.Studio.Bug.zip)

**To Reproduce**
Steps to reproduce the behavior:

1. Load file into Bambu Studio.
2. Select '0.28 Extra Draft' profile
3. Set initial layer hight to 0.28mm
4. Slice file
5. Inspect layer 89

**Expected behavior**
No extrusion across the model should be created.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: MacOS Ventura 13.1
 - BambuStudio Version 1.4.0.18

SaltWei, 2022-12-30:
This is corner case and bug caused by clipper lib.
Currently, you can rotate the model a little as workaround.
[image]

SaltWei, 2022-12-30:
This is a known issue and we are investigating

QingZhangBambu, 2025-05-30:
The issue has been resolved. Please download the latest version. Closing this matter.

closed 2025-05-30 by QingZhangBambu