bambulab/BambuStudio #980: Parts not properly aligned

toddanglin, 2022-12-29:


toddanglin, 2022-12-29:
**Extra context:**
- I'm working with multi-part objects exported from Fusion 360 as multiple, separate 3MF files
- I load the "base" 3MF file, and then attempt to add the "children" parts from separate 3MF files
- In the past, this would correctly position the parts relative to the "base"

**Work around:**
- If you export all parts from Fusion360 as a single, multi-part 3MF file and load that in Bambu Studio, the parts will be correctly positioned

SaltWei, 2022-12-30:
@zhimin-zeng-bambulab Please help to comfirm whether the behaviour is as expect

MadMax389, 2022-12-31:
Not working for me as well in v1.4 in a complex project with a lot of parts. It was working in v1.3.

zhimin-zeng-bambulab, 2023-01-10:
I try to reproduce this problem, but it doesn't seem to work. The picture below is what I did.
I found that the newly added object is always in the lower right corner of the original object.

[image]

zhimin-zeng-bambulab, 2023-01-10:
Could you provide your 3mf file, and the sub-part obejct file?  It will help us to solve the problem, thanks!

zhimin-zeng-bambulab, 2023-01-16:
Thanks a lot，we will fix it in future version.

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

Bathsheba, 2024-01-21:
This is not fixed. Added parts, and parts imported at the same time all drop to the build bed. They are not correctly positioned relative to each other.

SaltWei, 2024-01-22:
> This is not fixed. Added parts, and parts imported at the same time, all drop to the build bed. They are not correctly positioned relative to each other.

@zhimin-zeng-bambulab Please check and respond

zhimin-zeng-bambulab, 2024-01-22:
@Bathsheba  Could you share your 3mf file?  And could you use a picture to show your problem？

MadMax389, 2024-01-22:
I can confirm this behavior (v1.8.4.51). Image on the right, both stls loaded on the plate at the same time and assembled as one object. Image on the left, loaded car body first, then loaded window as a part...
[image]

Bathsheba, 2024-01-22:
> @Bathsheba Could you share your 3mf file? And could you use a picture to show your problem？

Sorry I didn't keep the file, it occurred in the context of trying to print a part in one filament completely enclosed by another part in another (translucent) filament. 
I was trying to create a part with a negative area floating inside it, and a second part occupying that area. I was not able to do this. Thanks for your consideration.

zhimin-zeng-bambulab, 2024-01-23:
@MadMax389 Could you share your stl files to help us solve this problem?

MadMax389, 2024-01-23:
I created a new example. 
This is a Fusion 360 drawing of two stacked cubes. Each body was exported separately. Cube stls are attached in zip file.
[image]

In this image on the left plate, both cube stls were added to the plate together and assembled as one object. On the right plate, the lower cube was added first and the upper cube was added as a part.
[image]

Prusa Slicer handles this operation properly,
[image]

[Cubes.zip](https://github.com/bambulab/BambuStudio/files/14017840/Cubes.zip)

zhimin-zeng-bambulab, 2024-03-27:
it has been fixed in v01.09.00.06

closed 2024-03-27 by zhimin-zeng-bambulab