bambulab/BambuStudio #981: Supports from separate bodies going into a body that's next to it

leofaoro, 2022-12-29:
**Describe the bug**
Supports from 2 bodies close together are being built inside the other body. See screenshot of first layer

**3mf File for This Bug**
[bug.3mf.zip](https://github.com/bambulab/BambuStudio/files/10322355/bug.3mf.zip)

**To Reproduce**
place 2 bodies close to each other (does not need to touch), enable supports

**Expected behavior**
merging of supports without interfering with the slicing of the adjacent body

**Screenshots**
[image]
this is showing the first layer

**Desktop (please complete the following information):**
MacOS Ventura
Bambu Lab Studio 1.4.1.52

MusangKingBambulab, 2023-01-01:
Hi, @leofaoro . I recommand to assemble these two objects together first (check 1st fig below) and then to generate support structure based on this single assembly (check 2nd fig below). If not, the support structure will be generated independently and overlaped just like what you have shown.

[image]

[image]

Since this is not the bug of this release. Please allow me to close this issue. Happy new year!

closed 2023-01-01 by MusangKingBambulab

leofaoro, 2023-01-01:
Fantastic thank you. I didn't know about the assembly part.