bambulab/BambuStudio #982: Send to sd card while printer is connected to different LAN

Ali229, 2022-12-29:
**Is your feature request related to a problem? Please describe.**
If you have multiple routers, you cannot send to SD card unless you are connected to the same network, you get the error as shown below:
[image]
Weirdly I am able to send it to the printer without any issues. I think this is a limitation as when sending to SD card it bypasses cloud, however this should not be an issue as all modern device are able to connect to one another as long as they belong to the same network. 
Example:
192.168.0.15 < Bambu IP
192.168.0.100 < IP for the 2nd router
192.168.1.20 < PC IP

**Describe the solution you'd like**
Able to send to SD card while the computer and printer are on different routers but still on same network like other slicers allow. (Sending to my other printers that are on 192.168.0.XXX through octoprint)

**Describe alternatives you've considered**
There are no alternatives, I have to wait for each print to end before I can send another one.

Merlwynd, 2022-12-30:
I can't send a file to the printer SD card even with only one router and one home wifi network. I have no trouble sending files for printing, viewing video, using Bambu Handy app, etc. I just can't send a file to the printer SD card. I get the same message as you got....that they printer and studio both need to be on the same LAN. How can they not be if I only have one router, and one home network? I ensured my computer (Win 10 laptop running Bambu Studio) was on 2.4ghz wifi, I have logged out and back in on both the printer and Bambu Studio. I have turned off my firewall for short time to see if that was blocking things....all with no change. I have tried asking for help on forums but did not get a lot of feedback...so I am extremely frustrated with this function as don't know what else to try.....wish someone could help me further....

scutler81, 2022-12-30:
Having the same problem. My home network is segmented into different VLANs including one for IOT devices which is the segment the printer is on. From my Windows 10 laptop I can't connect to the printer to SD card even when I try connecting it to the same WiFi network that the printer is on.

randyre, 2022-12-30:
Having the same issues due to using a Mesh network, all devices are the same IP address range but Studio says I'm not on the same LAN.

armmaker, 2023-01-01:
I’m having similar issues. I’m connected to my router via Ethernet and the printer is connected to that router’s Wi-Fi network. Printer and computer are both within the same IP address range and I’m getting the same erro as mentioned above that the printer and computer are not on the same LAN.

slurpey, 2023-01-02:
Same issue here - can send files to print without problem - get the "you have to be on the same LAN" for sending files to SD Card.  I do have a router to a Google Wifi mesh.  I see the live picture in bambu studio.  
[image]

As a note - by the nature of 3d printing and "makers", lots of us will be using complex network environments with iot devices, raspberry pi around, mutliple routers, coding etc.  So this will be a frequent need.  Thanks!

jup2000, 2023-01-03:
Something recently broke, and now I'm getting this error.  It's definitely on the SAME network (192.168.10.xxx).  

Signing in and out, and re-adding the printers no longer work.

vbkredeemer, 2023-01-06:
> 

Same here. 2 Accesspoints from Ubiquity and a Router from AVM. All meshed up. In beween a coax-bridge. So I am for sure in the same Network as my printer and "Send to SD-Card ist not possible. 192.168.1.*** . Hope this will be fixed. I am used to use klipper and this would be pain in the ... :-) Running up and down with a SD-Card is not longer state of the art.

clowrey, 2023-01-07:
Same problem here... Both printer and desktop in the same subnet... On Windows.

RaySajuuk, 2023-01-11:
Got my X1C today, same issue, same or different VLAN.

Ali229, 2023-01-22:
This is no longer an issue, there is a prompt now that allows you to manually enter IP address of your printer, just make sure you reserve that IP for the printer on your router so it doesn't change! :)

closed 2023-01-22 by Ali229

proNET-Southtirol, 2023-05-11:
sorry, but where ist this "prompt"? I am trying to send over to another subnet, it's not possible.

wilg, 2023-06-08:
Just bought the printer, cannot get the "send" feature to work at all. All the other connectivity features work, and the printer is definitely on the same network as the computer. Not sure why everything else would work fine except this.

[image]

slurpey, 2023-06-08:
> sorry, but where ist this "prompt"? I am trying to send over to another subnet, it's not possible.

Any news on this?