bambulab/BambuStudio #983: Make overhangs printable

mjr-woody44, 2022-12-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I wish the bambu slicer had the option to  “ make overhangs printable“ like cure. That is one feature that I use often in Cura for designs that don’t need full, supports just a little angle to make it printable.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add make overhangs printable to the slicer

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Nico04, 2023-02-15:
Same here, I use this option quite often in Cura, very handy 🙂
I'd love to have this on Bambu Lab Studio

zhichangxuexi, 2023-04-24:
[image]
Is this the function, as shown  yellow part  in the picture

odie5533, 2023-06-27:
Cura has this feature in their CuraEngine. They configure the setting as conical_overhang_enabled (which they show in the interface as "Make Overhangs Printable") https://github.com/Ultimaker/CuraEngine/blob/main/src/FffPolygonGenerator.cpp#L233

And the actual ConicalOverhang function is in https://github.com/Ultimaker/CuraEngine/blob/main/src/ConicalOverhang.cpp

I see this issue was assigned to @tangketan by @lanewei120. Is it being worked on? Being considered? Thanks.

Haidiye01, 2023-06-27:
ok,i have received your letter ,thank you!

Haidiye00, 2023-06-28:
[image]
Is this function in Cura what you need？

odie5533, 2023-06-28:
@Haidiye01 Yes, that's exactly it

tjc2005, 2023-11-18:
Really need this, orca has it but not bambustudio

tjc2005, 2023-12-12:
Come on, is this coming or not?

cbenhagen, 2024-02-11:
This has been implemented in this [PR in OrcaSlicer](https://github.com/SoftFever/OrcaSlicer/pull/1615) and might be ported to [PrusaSlicer in this PR](https://github.com/prusa3d/PrusaSlicer/pull/11967).

tjc2005, 2024-03-31:
I take it this very simple feature to port over... Is never coming then?

CombatVolcano, 2024-09-26:
I know this is a 'model issue' and these days people have generally learned how to design for printing, but not always. And even the text Bambu Studio generates will suffer from similar problems 

(image is modeled text not BS text)

[image]

Provodnik167, 2026-01-11:
Is there any hope that this feature will be implemented in Bamboo Studio?