bambulab/BambuStudio #986: Printstart with certain chamber temp

Merlinufti, 2022-12-30:
While Printing ABS o Nylon or something elfse with the X1C it is helpful to heat up the Chamber befor starting the print. There is a measurement for the chamber temperture. It would be nice, if the chambertemp could be set to maybe 45°C.

The solution might be heating up the heatbed, starting the Aux Fan. and if the temperature of the Chamber reaches 45°C the normal print procedure would start.

Thanks.

Kwintyy, 2023-01-16:
Would also love to see this feature. Or at least any type of chamber temp control.

attaway, 2023-02-07:
This would be amazing.  Currently printing ASA and would like to raise chamber temp before start

SaltWei, 2023-04-18:
Chamber temp of X1 is estiumated value rather than measured

aequasi, 2023-05-28:
Support for M191 would still be nice, even an approximate temp

simisoft-exo, 2023-10-08:
I'd love this feature too. Can't really print ASA or PC without warping if the chamber is too cold and I turn the heat bed on manually to 100C to achieve the temperature required.

GuzziRaz, 2023-10-10:
@SaltWei:
> Chamber temp of X1 is estiumated value

Why is that? The wiki says there's a chamber temp sensor on the button board. Is it too shielded / far away?

attaway, 2023-10-10:
> I'd love this feature too. Can't really print ASA or PC without warping if the chamber is too cold and I turn the heat bed on manually to 100C to achieve the temperature required.

I’ve had great results printing ASA without warping.
In addition to heating up bed prior I also cover in a foil insulation bubble wrap and I’ve been able to keep temps to at least 50 inside.

GuzziRaz, 2023-10-22:
I just confirmed that M191 is silently ignored.

With the upcoming X1S it obviously needs to be supported, and you really should enable it for X1C and P1S as well. Sure, we don't have a chamber heater but that means we need M191 even more - we have to wait for the *bed* to heat the chamber to *n* degrees before starting a print with eg. ABS. Also, at end of a print we might want to be able to use eg. `M191 R50` for running chamber fan until things are cooled down. Or for turning toolhead LED off once temperature is safe to touch.

If P1S doesn't even have a chamber temperature sensor, just use the nozzle temperature for M191.

Skye-Bambulab, 2026-11-16:
Hello, we've evaluated your feature request and consider it ppractical and feasible. It's now on our development roadmap and will be included in a future firmware release. I'll follow up here when the new version is available. Thanks for your suggestion!