bambulab/BambuStudio #987: Voron connection setup causes Bambu Studio to crash.

MandicReally, 2022-12-30:
On the latest version of BambuStudio, if I try to add a physical Voron printer and setup the wifi connection, Bambu Studio completely crashes immediately. 

**To Reproduce**
Steps to reproduce the behavior:
1. Select "Voron 2.4 300" from the printers list (or any Voron I have tested does it).
2. Click on the Wifi connection symbol next to the printer to add a physical printer.
3. Watch BambuStudio completely crash.

**Expected behavior**
Open connection window and allow customizing of printer name and add wifi ip address.

**Desktop (please complete the following information):**
 - OS: MacOS & Windows
 - Version 13.0 Ventura & Windows 10 20H2
 - Intel i5 2017 iMac
 - 2019 MacBook Air (Intel i3)
 - AMD Ryzen 9 PC

littleferris, 2022-12-30:
I am also experiencing this same bug. I get an unhandled unknown exception. followed by this error.
 `10:05:24 AM: BambuStudio got an unhandled exception: invalid map<K, T> key
10:05:24 AM: BambuStudio got an unhandled exception: invalid map<K, T> key
`
then Bambu Studio crashes.

- OS: Windows 10 
- AMD Ryzen 7

shafnitz, 2022-12-31:
Same issue here, only it's happening on macOS and not Windows for me. As soon as I click the Connection icon, the macOS application crashes immediately.

Amor-Roma, 2023-01-01:
Same here.  macOS Catalina 10.15.7, click wifi for Voron and it crashes

SaltWei, 2023-01-03:
@LiZ-Li-BBL Please help to check this issue. Thanks

LiZ-Li-BBL, 2023-01-09:
this crash has been fixed, and you can use the feature normally in the next release.

closed 2023-12-18 by LiZ-Li-BBL