bambulab/BambuStudio #991: Allow only one instance of Bambu Studio App.

Ali229, 2022-12-31:
**Is your feature request related to a problem? Please describe.**
If you have an STL already loaded in the bambu studio and want to load another one, you'd have to drag and drop it into bambu studio. Double clicking would open another instance of the app with the STL.

**Describe the solution you'd like**
Please add a option to allow only one instance of bambu studio just like other slicers.

julie777, 2023-02-22:
I often send models direct from Fusion 360 and I don't want a new instance of Bambu Studio to open for each one.

bambu123, 2023-03-02:
In Windows, it is a problem. So if you are working with a Windows system?

julie777, 2023-03-07:
I am working in Windows 10. Prusa Slicer has a preference "Allow just a single Prusa Slicer instance" show it should be easy to just roll in the code.

bambu123, 2023-03-08:
@julie777 
What can we do without the choice in the Preference Dialog?
Do you always respect the STL going to the previous window (or instance) with a double click?
This is the way that works on macOS currently.

julie777, 2023-03-19:
Remember that I am not using a double click to open Bambu Studio. Fusion 360 is running the slicer app.

I don't know of any way to make Fusion 360 use the existing window. It just executes the slicer application with a file parameter. Prusa Slicer somehow checks and sends it to the existing instance.

If you don't want to have a preference to only allow one instance why not just always restrict to one instance? In all my testing multiple instances of bambu studio is pretty much guaranteed to make studio hang. I suspect this has something to do with the network dll that it uses.

vpapanik, 2023-09-29:
That feature would be really useful for Fusion 360 !

magnumripper, 2023-09-29:
> In Windows, it is a problem. So if you are working with a Windows system?

I have the same problem under macOS.

MrDB42, 2023-10-17:
Having this would be great. Having Fusion 360 setup to use Bambu-Studio as the slicer program.
[image]

The workflow I use is:
1. Design part/component/object/model in Fusion 360.
2. In Fusion 360 select the part/component/object/model to print
3. In Fusion 360 go to "File", "3D Print", "OK" (see above image)
4. Do any adjustments in Bambo-Studio for printing
5. Slice model
6. Print.

Now if I want to add other objects to the build plate it would be nice to go back to Fusion 360 repeat step #3 and have the new part/component/object/model added to the build plate(s). A lot of times I have to go back and edit the part/component/object/model and I will just clear the build plate and would like to just repeat step #3.

Having another instance of Bambu-Slicer open up for every time I send a part/component/object/model from Fusion 360 to Bambo-Slicer is frustrating, slow, resource intensive on my computer, just to name a few. Plus, Fusion 360 is a VERY popular 3D design software, better integration would be very beneficial to those of us who design stuff.

jbkuma, 2023-10-20:
+1 on this, it's sort of a critical feature and I already miss using Prusa Slicer now that I've picked up the P1P!

This was native to the slicer prior to the fork, please bring it back!

Andre-KVAGeradores, 2023-11-29:
+1 here. Works single instance in MAC OS, but not in windows. Multiple instances is a pain...

JustinKScott, 2023-11-29:
Agreed!  I really find this frustrating.  I miss the workflow from my MAC.  

I typically have multiple models being printed on the same sheet.  

Please @bambu123, can we have this feature?

andysmithfal, 2023-12-12:
+1, I've recently switched from PrusaSlicer to the Bambu ecosystem and this is incredibly frustrating for the Fusion 360 workflow

shichao402, 2023-12-18:
我是新用户, 我在windows,mac, andoird三个平台上尝试使用了stuido或者handy, mac的体验最好, windows平台上最让我难以忍受的是每次从makerworld打开模型都会产生一个全新的studio实例. 希望改进: 当从makerworld打开模型时, 不要总是打开新实例. 让我自己立刻选择, 或者预设一个开关是否总是打开新实例

I am a new user, and I have tried using Studio and Handy on three platforms: Windows, Mac, and Android. The experience on Mac is the best. What I find most intolerable on the Windows platform is that every time I open a model from MakerWorld, it generates a new instance of Studio. I hope for an improvement: when opening a model from MakerWorld, don't always open a new instance. Allow me to choose immediately, or preset a switch to decide whether to always open a new instance.

MrDB42, 2023-12-19:
Just an update on my personal workflow... I gave up on using the "3D Print" function in Fusion 360.

Now I export my models as STEP files, then bring the STEP files into Bambu Studio (i.e. no more STL files). I'm finding that Bambu Studio does a better job slicing STEP files than STL files. I don't know what "voodoo they do" for the coding, but I'm getting better prints with STEP files.

hkdsun, 2023-12-28:
Just migrated from PrusaSlicer and having the same issue

CROKOBACON, 2023-12-31:
+1 same issue, my workdflow now is terrible and i have my folders flooded with unfinished stl (before i was just send from F360 to PrusaSlicer in the same instance of the program, now i need to export an stl and manually import to have all in the same instance of Bambu Studio.

a42wg, 2024-01-13:
+1 current behavior is quite annoying

IonNight, 2024-02-18:
+1 only allow one instance of Bambu Lab slicer isnt really the issue, but I need a way to import files from F360 without have to deal with a new instance all the time. Several instances are good to have when using printers from different brands.

There is a way to make F360 import STLs directly into PrusaSlicer with a bat file - found here: https://github.com/prusa3d/PrusaSlicer/issues/5420?fbclid=IwAR1AN3-1rjvkx5Q4tiray-5-0mHqS1XVVJtJdfpnTipY9T9V5OEe11LxUYs

I've tried to use the bat for bambu-studio.exe too, but it doesnt work

mysterious-roman-dodecahedron, 2024-03-24:
I am finding that clicking the button "open in bambu studio" on makerworld also opens a second instance on Windows 10, even if I didn't have a project or model loaded in my original instance (just monitoring the printer).

[image]

[image]

Virrjet, 2024-04-21:
+1. This is very annoying.

OzDruiD, 2024-04-21:
@Virrjet @mysterious-roman-dodecahedron @a42wg 
So - for those needing a  Single instance, and not imports and the F360 use case etc, there is an option in Studio, at least in my latest  version, that does exactly that.   

For your info - Open Preferences, under the language and measurements "**_Keep only one Bambu Studio instance"_** is an available check box.....

andysmithfal, 2024-04-21:
@OzDruiD you're right! I had to update to the 1.9.0 beta to see this option, but once enabled it performs exactly as expected. Exporting from Fusion 360 will import the file into the existing instance of Bambu Studio. 
N.B. installing the beta version erased my local data/preferences, if you don't sync to Bambu Cloud you will definitely want to either hold off or take a backup first.

Virrjet, 2024-04-22:
> @Virrjet @mysterious-roman-dodecahedron @a42wg So - for those needing a Single instance, and not imports and the F360 use case etc, there is an option in Studio, at least in my latest version, that [...]

> @OzDruiD you're right! I had to update to the 1.9.0 beta to see this option, but once enabled it performs exactly as expected. Exporting from Fusion 360 will import the file into the existing [...]

Thank you both!

Lancaban, 2024-05-11:
I came here from an Internet search, was having a similar problem, I'm running Linux, and using the "open in Bambu Studio" button on maker world starts a new instance each time. 

This is not the world's worst bug, it's more annoying than anything, but I thought I'd share my input that the problem exists on Linux as well (at least for me).

isycat, 2024-11-19:
**1.** PrusaSlicer has the CLI parameter --single-instance which does exactly this. 
I can see references to single-instance param in the codebase but it does not seem to work (Windows; Bambu Sudio 1.10.1.50)
`"bambu-studio.exe --single-instance"` instantly exits with no output and code 0
The same is true of `"bambu-studio.exe --no-single-instance"` and `"bambu-studio.exe --random-garbage"`

**2.** PrusaSlicer also uses the URL protocol "prusaslicer://" to ingest stuff from printables.com.
Switching the URL protocol to open Bambu Studio results in it launching with "The file does not contain any geometry data"
I am not clear why the same feature in BambuStudio passes the data ddifferently, but it feels like the upstream* method should be supported as well.

It feels as though support for these 2 things should be pulled in from PrusaSlicer :) 
  (and that'd be really great!)

jbkuma, 2024-11-19:
This has been implemented in the latest versions and it appears to work as expected.
[image]

Also implemented in Orca Slicer
[image]

isycat, 2024-11-19:
The checkbox works, but the **CLI param is broken**.

URLs other than makerworld also do not work.