bambulab/BambuStudio #992: Multiple Filament Slots With Same Color (with AMS)

MadMax389, 2022-12-31:
**Describe the bug**
If multiple filament slots in the slicer are filled with the same color/material, Bambu Studio maps the first color regardless if the color in a higher slot is selected on the model. For example, if white PLA is selected for both slots 1 and 2, and I want the printer to use the PLA in slot 2 first (remnant roll), then switch to slot 1 (full roll) using the AMS auto switch function, BS will map the print to slot 1 instead of slot 2 when sending to the printer. It is possible to remap the slot on the send screen and the AMS switch function works properly, but it is easy to miss if not paying attention.

**3mf File for This Bug**
Not required

**To Reproduce**
Steps to reproduce the behavior:
1. Add a cube primitive to the plate
2. Set the color on filaments 1 and 2 to white generic PLA
3. Select the cube to be printed with filament 2 and slice. (shows proper filament in the slice overview)
4. Send print
5. Print is mapped to slot 1 

**Expected behavior**
Print should be mapped to slot 2

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - BS v1.4.0.18

Benderrodrigu3z, 2022-12-31:
you can click on that image that you circled and tell it A2, as for defaulting to another line that might be a bug

MadMax389, 2022-12-31:
@Benderrodrigu3z Yes, remapping on the send print screen does work, and it switches properly. I missed the mapping the first time I did it and wondered why it printed from the full spool and not the remnant spool. Now that I'm aware, I can make sure it's mapped properly but it seems like given the choice between two slots with the same color it should default to the filament selected in the slicer for the object.

toddanglin, 2022-12-31:
I'm not sure I'd consider this a "bug," but perhaps a feature request?

You can arbitrarily configure filaments in Bambu Studio without any consideration for what is physically loaded in your AMS units. When it's time to send your print, Bambu Studio will make a "best effort" guess to match your Studio filaments to physically loaded filament in the AMS -- first matching material type (PLA, PETG, etc) and then color.

That said, I have seen other suggestions that Bambu Studio add an ability to populate the filaments in Bambu Studio based on what's actually loaded in AMS units. If that ability existed, then it would make more sense for Studio to "remember" the Studio <> AMS mapping.

Benderrodrigu3z, 2022-12-31:
you can already do that @toddanglin, you have to hit the pull request button but to auto do it might be smart but that would have to be a push function from the printer.
[image]

MadMax389, 2023-01-01:
_That said, I have seen other suggestions that Bambu Studio add an ability to populate the filaments in Bambu Studio based on what's actually loaded in AMS units. If that ability existed, then it would make more sense for Studio to "remember" the Studio <> AMS mapping._

@toddanglin As Benderrodrigu3z mentioned, you can sync the actual filament color/material into the slicer. I always do this rather than set the filament color in the slicer since exact RGB values will be synced. Even when colors are synced from the AMS, the slicer still chooses the first slot with the model color rather than the filament selected. It's not a major issue and I suspect the solution is a little more complex since the slicer seems to match colors rather than slots on the AMS.

[image]
[image]
[image]

toddanglin, 2023-01-13:
Thanks for the pointer to that feature! I missed that the option to sync with AMS was already available.

I'll add that I do appreciate that filament configurations are saved with Bambu Studio projects and not dependent on what I have loaded in AMS. Makes it easy to switch between projects that use wildly different AMS filament load-outs without worrying about the physical AMS mapping until it's time to print.

But as you say, it's likely a complex heuristic.

CameronBigger, 2023-02-02:
I'd agree this is a bug from the user's perspective as per the OP. After all, the slicer confirms that AMS slot 2 filament will be used, then at minimum it should **default** to that on the "Send print job" interstitial dialog. 

I too always always sync the colors/filaments from the AMS to Bambu Studio before I do the final slice and print. Reduces surprises, especially if there's the same color as I have been planning in BS in the AMS, but I'd forgotten that I switched out PLA for say PETG previously. So after syncing colors/materials from AMS to BS, I do the slice and get the warning that my cool plate setting isn't compatible with the PETG...which I had forgotten was in the AMS.

Sync feature is "pretty good" but it still fails occasionally (or only replaces a few of the 4 filaments when it should replace all). Would like to see it be a little more robust and bulletproof but it works 95% of the time as it should in my experience.

bobtitus, 2026-09-28:
Same problem here, on an H2S with AMS 2 Pro.

Setup: A1 and A2 both loaded with black ABS (third-party spools, same preset and colour).
The object is explicitly assigned to filament 2 in the slicer, which corresponds to A2.

Result: the send dialog always proposes A1. The print only uses A2 if I change it
manually in the send dialog. Disabling "AMS filament backup" makes no difference.

Tested with Bambu Studio 2.7.1.62 and 2.8.2.61, and OrcaSlicer 2.4.2, same behaviour in all three.

When I assign a specific filament to an object, I expect that choice to be kept.
At minimum, please keep the explicit assignment when a slot has been chosen,
or show a clear warning when the mapping differs from what was selected in the slicer.
It is far too easy to miss, and it means using a spool I did not intend to use.