bambulab/BambuStudio #995: Studio can no longer communicate with printer after stopping a print

akbyrd, 2023-01-01:
**Describe the bug**
If I manually cancel a print using the LCD screen on the printer, Studio can no longer communicate with the printer. I cannot view the camera and cannot start a new print.

Studio says the printer is busy with another print, even though the print was cancelled. I have to power cycle the printer and wait a minute for it to start working again (indicating it _might_ be a firmware issue).

Related: I cannot cancel a print from Studio. Nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print
2. Cancel the print using the LCD
3. Attempt to start a new print from Studio
4. Notice the new print cannot be started

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 22H2
 - Version 1.4.1.52

jasonmooney, 2023-01-04:
I also ran into this my first day of printing with the P1P

lanewei120, 2023-01-05:
@jasonmooney @akbyrd 
thanks for reporting this issue
it seems the state sync not working correctly

could you help to fetch the logs from the printer's sdcard and upload here?

also please helo to fetch the studio following this guide, and upload here too.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log