bambulab/BambuStudio #997: 0.2 profiles ( and possible the other ones )

PowerChaos, 2023-01-01:
**Describe the bug**
The 0.2 profile is a big mess , the speed settings are extreme low
but the biggest problem are the system profiles , they contain value's of 1mm³ and 2mm³
so that makes those profiles useless

**3mf File for This Bug**
Not related to 3MF, it are system profiles

**To Reproduce**
Steps to reproduce the behavior:
1. select the 0.2 profile
2. Check max flowrate

**Expected behavior**
atleast the same value's as the default profile , 

**Screenshots**
not needed

**Desktop (please complete the following information):**
 - OS: windows
 - Version: 11

ashleyuk90, 2023-01-01:
What printer are you seeing this for?

Are you seeing it under the filament settings or somewhere else, from what I can see in my Bambu Studio, the only setting for Max Volumetric speed is under the filament settings which mine for Bambu PLA Basic shows 21.

And checking the 0.2mm profiles for 0.4 X1C mine is showing in the 100's as per expected.

PowerChaos, 2023-01-01:
[image]
thats the default profile for the 0.2 nozzle profile
i am not talking about the 0.2 layer height , i am talking about the Nozzle 
i edited my tittle and post as i forgot to mention that it was the nozzle profile 🤦‍♂️

c2h5oh, 2023-01-02:
@PowerChaos while we're waiting for this to be fixed would you mind manually increasing max volumetric speed and printing this https://www.printables.com/model/318350-bambu-slicer-max-flow-test-tower-for-x1c to get actual value we want to use?

PowerChaos, 2023-01-02:
[small nozzle template.zip](https://github.com/bambulab/BambuStudio/files/10331745/small.nozzle.template.zip)
@c2h5oh 
this is the settings i use now
for everything else i just use the default profile as the 0.4 profile
as every setting is "broken" , defaults go to 50 mm/s and 150 mm/s is max 
so to make a real profile/test i need to change a lot and spend some time on it

Please note , i only edited the GENERAL profiles , not the Bambu or other brands

i personal think it is better to leave that part to the engeneers to create a default profile as they know what value's it should be designed for 

for refference here are my current settings and they work great in combination with the flowrate settings in above profiles
[image]

MadMax389, 2023-01-02:
I just bought a .2mm nozzle and was wondering why my print was going to take 7 days! Thanks for the workaround @PowerChaos

MadMax389, 2023-01-02:
Could you add the .2mm printer profiles to the "compatible printers" section of Generic PLA.json file".

Example:
{
    "type": "filament",
    "setting_id": "GFSL99",
    "name": "Generic PLA",
    "from": "system",
    "instantiation": "true",
    "inherits": "Generic PLA @base",
    "compatible_printers": [
      "Bambu Lab X1 Carbon 0.4 nozzle",
      "Bambu Lab X1 0.4 nozzle",
      "Bambu Lab X1 Carbon 0.6 nozzle",
      "Bambu Lab X1 Carbon 0.8 nozzle",
      "Bambu Lab X1 0.6 nozzle",
      "Bambu Lab X1 0.8 nozzle"
      _"Bambu Lab X1 Carbon 0.2 nozzle",
      "Bambu Lab X1 0.2 nozzle"_
    ]
}

PowerChaos, 2023-01-02:
tried it , it does not work that way (in my case ) 
as then the program refuse to start up , it trows out a error saying that profile .2 is missing and it crashes 
atleast in my case , atleast there is a small workaround for now

SaltWei, 2023-01-03:
This is not a bug. 0.2 nozzle can't print with high Max Volumetric speed which is same with 0.4 nozzle.
0.2 nozzle is VERY easy to be clogged with high Max Volumetric speed.

SaltWei, 2023-01-03:
No matter how much speed is set in process preset, the final speed will be limited by Max Volumetric speed in filament preset for protection.

PowerChaos, 2023-01-03:
> This is not a bug. 0.2 nozzle can't print with high Max Volumetric speed which is same with 0.4 nozzle. 0.2 nozzle is VERY easy to be clogged with high Max Volumetric speed.

so if i understand it correctly 
the flowrate get capped at 1 mm³ ? thats even slower then tpu speed on 0.4 

so far my testing it works fine , i indeed suffer from clogs a lot ( at the flowrate of  0.4 profiles ) but those also happened on the 0.4 nozzle for me , so not sure if it is the flowrate as the prints looks nice  xD

but i still feel that 1mm³ is way to low to be any good , it should be atleast 8 or so ( from 21 to 8 seems do able , or from 16 to 12, got no idea about great value's or i need to test them out)

KD7LRJ, 2023-01-03:
I just wanted to mention that the maximum flow rate set in the PLA 0.4mm stock profile for the X1C/P1P is not realistic for most circumstances and you shouldn't base your expectations for a 0.2mm nozzle on those settings.

c2h5oh, 2023-01-03:
@SaltWei 2mm^3/s sounds awfully low. For 0.4 nozzle Bambu PLA Basic has max volumetric speed of 21 mm^3/s. With 0.2 having 1/4 of the area of 0.4 one would expect that number to be around 5 mm^3/s.

SaltWei, 2023-01-05:
It's open for user to edit and change the value.

This value is tested and changed by our tester for many models and printers and at least safe enough.

SaltWei, 2023-01-05:
Mark as enhancement of preset and will transfer this message to filament guys.

c2h5oh, 2023-01-06:
> This value is tested and changed by our tester for many models and printers and at least it's safe enough.

@SaltWei Would it be possible for your filament test team to share at least some of their testing workflow? I've got all nozzle sizes and about 50 different filaments from a selection of brands and I'd love to be able to create and share some more presets for community to use.

SaltWei, 2023-01-07:
> @SaltWei Would it be possible for your filament test team to share at least some of their testing workflow? I've got all nozzle sizes and about 50 different filaments from a selection of brands and I'd love to be able to create and share some more presets for community to use.

Response from filament team and they will prepare one wiki page for this.

MadMax389, 2023-02-26:
I checked the standard Prusa profiles for the Mini with the .25mm nozzle on a couple of models. The max flow rate with standard settings was less than 1.5 mm3/s. I bumped up the speeds to get the flow rate up to 4 mm3/s and the nozzle clogged pretty quickly. Bambu's defaults may not be too far off, but I do think the user should have discretion to raise the max flow rate if desired.

GuzziRaz, 2023-10-02:
> Response from filament team and they will prepare one wiki page for this.

Did they ever do this? I can't find it.