bambulab/BambuStudio #1001: Send to Printer stating "not same LAN" when printer IS in same lan and can be connected to

Lesani, 2023-01-02:
**Describe the bug**
[image]

**Expected behavior**
sending to the printer works from bambu studio

**Desktop (please complete the following information):**
 - OS: [image]
 - Version [image]

psiberfunk, 2023-01-02:
Bambu has silly hardcoded masks for what they think a LAN is .  If it’s not 192.168.X.X they don’t treat it as on a lan .  Not sure why they do this , but this is a firmware bug I think .

Merlwynd, 2023-01-02:
Have exact same issue as Lesani …everything works right except sending file to printer SD card. All my network addresses are 10.0.0.xx also. First I have heard about the 192.168.x.x requirement…thanks for that. But there are folks that have problem with that as well. See issue #982. I have seen quite a few posts (and added to them) about this “not on same Lan” issue but have seen no reply from Bambu. As I understand it 192.xxxx is older architecture vs 10.xxx?….but must admit I am not that knowledgeable in this area. I have even tried the “Lan Only” mode on the printer…but get the same message.

xcschultzx, 2023-01-02:
Same issue here, 10.10.1.x network. It has been driving me crazy. Makes sense this is a problem if they require a 192.168.1.x network. Please fix this issue Bambu.

psiberfunk, 2023-01-02:
It has nothing to do with architecture and everything to do with most commonly used.  Most consumer routers default to the 192.168.X.X scheme , but some do not .  They’re all valid private LAN subnets .  This is probably someone just being really sloppy in the code on windows and refusing to listen for ssdp on the right interfaces .

What it seems to do is that it only even opens a socket (udp 2021, ssdp) to listen for printers existing on interfaces with a 192.168 address

Merlwynd, 2023-01-02:
Thanks so much for reply on this… as I said my knowledge is limited in this area so really grateful for your knowledge…I am never too old to learn…LOL! Googling around it seems the commonly private ipv4 address networks are…
iP 192.168.0.0 - 192.168.255.255 (class A network)
iP 172.16.0.0 - 172.31.255.255. (class B network)
iP 10.0.0.0 - 10.255.255.255 (class C network)
So, Bambu should ensure any of these address ranges can constitute on being on same LAN (i.e. allowing for the various correct listening ports?)…do I have it right?
Unfamiliar with listening port being used in my case but can I guess I find that out using cmd line ….though I assume I cannot simply change it to the one you mention while still being on iP 10.0.0.xxx? My cable company consumer router must default to that network as I did no changes (wouldn’t know how)..and I have nothing complex setup…just one home wifi network…

Lesani, 2023-01-02:
no, Bambu should check if the IP address of the printer is reachable and is indeed the respective bambu device, no matter what IP address/range is used.
Next week I will put a printer on our company network, and of course it will not be 192.168... but I will need local access because of data security, so I'm back to FTP there as well...

psiberfunk, 2023-01-02:
> no this should not be limited to ANY range, Bambu should check if the IP address of the printer is reachable and is indeed the respective bambu device, no matter what IP address/range is used.
> Next week I will put a printer on our company network, and of course it will not be 192.168... but I will need local access because of data security, so I'm back to FTP there as well...

The printer should probably never be on a public IP , therefore they should allow all non public ip subnets .  All ips doesn’t make sense to listen on

rkarlsba, 2023-01-02:
That a unit is on a public IP address or not, doesn't mean if it's available from the net or not. There are such things sa firewalls, after all. This reminds me of the bug about not being able to add a printer by its IP address. Several home networks have different VLANs for different units these days, after all.

psiberfunk, 2023-01-03:
So , I think the problem here is it is the SLICER doing the listening .  And if the IP your slicer is on is on a possibly publicly accessible IP, it would be bad security practice for it to be listening on that IP .

rkarlsba, 2023-01-03:
I've only workd with IT since 1994 or so, but for what I know, making the slicer listen to what the printer has to say, is a very bad move. The printer's role is a server, the slicer is a client. Better just setup this in a secure, encrypted way (meaning no FTP, as I wrote about in https://blogg.karlsbakk.net/2022/11/23/bambu-lab-x1-carbon-the-flipside/), but something good. https or sftp should work, but using Matter as a new protocol, will secure it all.

psiberfunk, 2023-01-03:
> I've only workd with IT since 1994 or so, but for what I know, making the slicer listen to what the printer has to say, is a very bad move. The printer's role is a server, the slicer is a client. [...]

So , it’s not really that insecure if you’re using it on local LAN mode .  Self signed HTTPS certs don’t magically get accepted and would likely cause as many issues as they create .  The UDP based service discovery they use is somewhat typical , although is not related to the security of the FTP server .  If you’re really worried about security on your local LAN, you’d want to setup a dedicated VLAN, which would inherently give you that level of assurance .  In any case I think this is getting a bit off topic here .

rkarlsba, 2023-01-03:
Good luck adding the printer across a router as you would with a separate VLAN. I posted this https://github.com/bambulab/BambuStudio/issues/702 aboutadding printers by IP address or DNS name and I beleive that's the way to go.

walterwongbbl, 2023-01-09:
This error tip is not accurate. We will allow users to enter the printer's IP address in the next version.

rkarlsba, 2023-01-11:
It is indeed accurate until you release this version ;)

Lesani, 2023-01-19:
fixed with latest release, sending works now

closed 2023-01-19 by Lesani

intdel, 2023-05-12:
> This error message is not accurate. We will allow users to enter the printer's IP address in the next version.

Did this ever happen? I am having the same error.

zswickliffe, 2023-05-20:
@Lesani I'm still having an issue with this. I've enabled all ports listed [here](https://wiki.bambulab.com/en/general/printer-network-ports) on my PC firewall settings as well as in my router. I'm not sure what else to try - any tips?

tyxdqq, 2023-05-31:
> i hade the same issue 4 mins ago, so i turned off my windows firewall, private net firewall, now it works......
i just turn on my domain firewall, and it still works, so i guess the private net firewall it is

zswickliffe, 2023-05-31:
> 

I resolved the issue by creating an inbound and outbound firewall rule for Bambu Slicer allowing all ports. 

I discussed with support and they didn't know about this, I ended up resolving it on my own.

walterwongbbl, 2023-06-05:
Hi. zswickliffe
Thank you very much for your suggestion. I will verify if your method is feasible.

Znubbis, 2023-08-29:
By reading this thread it sound like Bambu Labs are not ready to enter the corporate sector and they will continue to be a deluxe home printer. Most corporations do not have 192.168.x.x as a internal LAN address or have only one subnet. They do probably not allow wifi clients with only a password or allow clients to access the internet with several ports other than 80 and 443.

Bambu Labs need to add a option to enter a IP in their software to access the printer as its rare their automatic detection will work everywhere and create a real "LAN Only" mode one that actually work.

I'm disappointed as we just got a printer at work as this is one of the only real fast printers.

xcschultzx, 2023-08-29:
"Most corporations do not have 192.168.x.x as a internal LAN address"

This was fixed and closed quite a while ago. I have a 10.10.1.x network and I can print locally no problem. You always have the option to print from SD card if you need to be completely air gapped.

BenDavidson90, 2023-10-02:
As of today, I can start the print without any problem but I can't send file to printer just to print later.
It says "The printer is required to be on the same LAN as Bambu Studio". it's incomprehensible since if I send for direct printing it finds the printer...
Did the Windows Defender trick but nothing change.

tyxdqq, 2023-10-02:
same thing here . it been like this for a month now, tried pretty much
everything LOL
i gave up~ just print directly from the printer for now

walterwongbbl, 2023-10-07:
Hi. @BenDavidson90 @tyxdqq @xcschultzx 
First, this error message is not accurate. When we can't retrieve the device's IP address, we will display a message saying that it's not in the same local network.

In response to this situation, if we can't obtain the printer's IP address, we will display an input window, allowing the user to enter the device's IP and access code.

We will incorporate these changes in the next version.

Lesani, 2023-10-07:
You can just determine its IP address by asking your cloud service, and only if ‚lan only‘ is checked fall back to your offline methodSent by my iPhone

EinzelFall, 2023-10-18:
As someone that doesnt know anything about network and this stuff, i dont know what to do now. Have the same issue as BenDavidson90.
[image]

> As of today, I can start the print without any problem but I can't send file to printer just to print later. It says "The printer is required to be on the same LAN as Bambu Studio". it's [...]

Also did the firewall thing, didnt change anything. Also have one of those 10.10.1.x networks. What exactly am i supposed to do now, so i can send the files to my sd card?

Thanks in advance

snoopyapocalypse, 2023-10-25:
I am having the same issue. I have xfinity so I have ip addresses beginning with 10.
I even tried separating out the networks into 2.4 and 5ghz and making sure BS and the x1c were on the 2.4. Then that didn't work I was out of ideas and searched to find that this is probably a bug.

judge2005, 2023-11-12:
This is still happening. I'm on a Mac and the addresses are all 192.168.1.x

Siggy101, 2023-11-23:
I have just received my X1C combo today, and am really annoyed to also be getting this issue.
I'm on the most ordinary home network, I tried the exceptions in Windows Defender, have a 192.168.x.x range, etc etc etc
I get this same error in Bambu Studio and Orca Slicer.  
I have opened a support ticket with Bambu so let's see what they come back with.
(for the record, I also had the annoying blank login window issue too.  I fixed that one by reinstalling and forcing the network connector install).  I have left Creality as I thought Bambu was turnkey...........

Siggy101, 2023-12-02:
It's finally here guys!
I raised a support ticket for this 'not on the same LAN' bug. 
Bambu tried all sorts of things that didn't work, but then they told me to install the beta 2 from this link and it worked!!!
https://github.com/bambulab/BambuStudio/releases

OK, it's in beta for now, but when I tried to send a file to the printer, this build prompted me to input the IP address of the printer, as well as to validate the access code, and then it worked perfectly.

Fill your boots guys!  Go try this beta and report back if it fixes the issue for you.  It certainly has for me.

StephanBieker, 2024-01-06:
How often did it work? I'm stuck with 1.8.2 and it doesn't work.

StephanBieker, 2024-01-06:
Very annoying bug