bambulab/BambuStudio #1002: Pls allow to combine VASE mode and normal print mode / different objects

DM4DS, 2023-01-02:
Problem: I´m coming from a commercial slicer and there I am easily able to combine VASE mode and "normal" print mode in one object by spliting the job in 2 sections.
This is not possible in Bambulab slicer. Tryed to "cheat" with multicolor, spliting objects , loading 1 part for the vase mode job + 1 for the normal printing but the slicer drops the second, which should be at height x, to the build plate and still I´m not able to enable vase for one and normal for the other.

Solution: Alow, when splitting the part in 2 objects (or more) to enable vase mode in one and doing a normal print job in the other(s).

My alternative:
I´m exporting gcode right now and edit 2 files into 1, copy it on the sdcard and print the job, but that´s annoying.

Content:
You can watch my "help" video on video. There you also see the printed part from an older , not optimized profile, combined vase/normal printing, yes there are parts were this is extremly helpfull and safes a ton of time and material:
https://youtu.be/qdB6qTi6eeA

barrettj, 2023-01-05:
I don't know if you'll be able to continue on with the vase mode part, but you can raise an object if it's part of an assembly

If you select both objects and "assemble" them, you can then choose just one part form the object list and raise it.

[image]

[image]

mkosmo, 2023-01-07:
Vase mode does a continuous spiral instead of the discrete layers of non-vase mode, so I'm not sure how the two modes would be compatible simultaneously.

DM4DS, 2023-01-07:
Thanks for reply.Look into simplify3d , there it is easy to do and I print parts since years in this combination.Can send you videos....GreetingsSandor Dubronner -- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

Malitybo, 2023-10-04:
I agree this would be a great addition. I use it to create lampshades. I use the bottom layers for a sturdy base of the shade from which it hangs and then the follow-up layers in vase mode which are for aesthetics. I still use this on my ultimaker in combination with simplify3d, but would love to use my bambulab printer for these jobs. Any clues if this feature will be added somewhere soon?

michaelwschultz, 2024-02-13:
+1 Would love to see this as well. having the ability to use vase mode starting at a specific layer can really help with print times as well as quality of certain prints. Hope this feature makes it into the pipeline soon.

ndonathan, 2025-07-05:
I'd love flexibility in being able to do this, but just being able to transition from layer printing to vase mode at a specific layer would work in all my use cases. I wouldn't need the ability to switch modes more than once and vase mode would always be on the upper layers.

m9tdev, 2025-08-03:
I've found a solution/workaround for this problem (as long as the first part fits under the print arm when the toolhead is on the plate): I wanted to print a base with PLA Basic and a wall on top of it using PETG Translucent using Spiral Vase mode.

Step 1:
Make sure the object is in a place that doesn't interfere with toolhead cleaning and homing (e.g: bottom-right, as long as the first object fits below the print arm, otherwise this probably doesn't work).

[image]

Step 2:

Select `Objects`, select the part you want to print on top later, and move it below the plate.

[image]

Now you can print that part with the settings you like.

Step 3:

Revert moving the part below the plate, and instead now move the first/base part below the plate.

[image]

You'll see the rest now floats above the plate

[image]

Now print that part with the settings you like (e.g: Spiral Vase mode)