bambulab/BambuStudio #1003: Using Folders on the LCD Screen on the SD Card

Maximizer18, 2023-01-02:
Hi, I'd like to request a feature added where you can put things in folders on the SD card, and see the folders on the LCD screen. I like to stay organized, and that way I know which prints are for which nozzle, material type, multi color, and so on, without naming each and every file what it is. I can just put it in that folder that has only PETG prints, or only PLA prints, and so on.

Lesani, 2023-01-03:
yes please, I also keep many files on the SD for quick access directly at the printer, and on my prusa machine I can store them by my folder structure. I do store them in a structure on the bambu already, in the hopes of this actually being implemented at some point

RobinWhitfield, 2023-01-04:
Definitely add me to the list of people desiring folder support.

vbkredeemer, 2023-01-06:
Here too. Would also be great to have access to that folders to manually add gcode-files. More like klipper. But I see this maybe is the "media"-stuff for X1 and X1C and only for that printers?! Just to have folders, you can use from bambuslicer when sending a file will be a big improvement.

FreeManFreeThinker, 2023-01-08:
I would love to be able to do the following:

- View/open folders and files in bambu studio that are on the sd card while it's in the printer. Being able to open 3mf files on the sd card, view the project natively in Bambu studio, make changes, and re-save with a different name would be insanely awesome.

- Modify and delete folders/files on the sd card while it's in the printer.

- More control over the 3mf file preview image. Simply add the ability for us to adjust the angle at which the image is taken. Sometimes the thumbnails for the files are taken at an angle that doesn't show all the print/objects on the plate.

CCS86, 2023-01-12:
Another vote for folder support!

Also, being able to view and modify internal storage and SD storage via BS would be very helpful. the touchscreen on the printer is quite laggy.

johnmbaughman, 2023-01-17:
+1 for this!

sodachen22, 2023-03-10:
Thank you for the feedback. An internal requirement id (BBL001-5079) is created, this feature will be evaluated and planned.

Maximizer18, 2023-03-10:
I see this may be implemented, which is awesome! I have purchased a P1P since I first made this request, and would love this same kind of feature to be able to be used on the P1P as well. Obviously no touch screen, but being able to navigate folders on the screen we have on there would be an awesome feature! Thank you!

Benderrodrigu3z, 2023-03-31:
> I see this may be implemented, which is awesome! I have purchased a P1P since I first made this request, and would love this same kind of feature to be able to be used on the P1P as well. Obviously [...]

it has yet to be added

Maximizer18, 2023-04-02:
> it has yet to be added

I know, but they said it's being "evaluated and planned"

homewerken, 2023-06-02:
would like to push it again. ore is it still possible?

lucian151, 2023-10-26:
We cannot express enough how important this feature is for us. We have over 100 files we need to browse through in random order each time we need to kick off a print. This is multiplied across the 6 machines we have.

mmuellerphoto, 2023-11-07:
It's really annoying to scroll through all the files ... folders please!

realmadridsi, 2023-11-10:
Can't stress enough how important this is for people like me who use their printers in a print farm. People running print farms have too many products to have them all listed in random order.

claytonhoxworth, 2023-11-19:
Yes! I am waiting for this as well. My experience running my X1C's would go so much smoother if I could store print files into folders and see the folders on the machine. I have over 170 files that I can sort through and find with no issue on other machine brands with folder support, but it's a pain on my 7 X1C's! Please Bambu add this!

iliya-yordanov, 2023-11-21:
+1 for folders.

napobonaz, 2023-11-23:
Another +1 for folders (multiple X1 printers). 
Any status update on this feature? Would be really really helpful.

KytorIndustries, 2023-12-11:
Any update on this? @sodachen22

koder-guy, 2023-12-16:
Looking at moving from clipper/mainsail were we had folders and it was great. Really disappointed to find no folder support on Bambu printers?? How is this not possible??. I can't work without organising all my files into folders so without this I will not be purchasing any more Bambu printers. It makes using the printer unbearable.

firkateyn82, 2023-12-25:
+1 for this function. I can not understand the point of not having folder support. What do they expect having 100s of files in the root?

fabiomaticus, 2024-01-20:
News for this request? I

CCS86, 2024-01-20:
Please please please!

mackbon401, 2024-03-22:
+1 for this! I run a university makerspace and we have 15 XC-1 printers on their way. There are hundreds of students that come through the lab to print things. I'm really hoping this gets resolved by August when we officially open our new space, otherwise it is going to be hectic.

Even our crappy Lulzbot printers have folders!

Sissou7, 2024-06-05:
+1 please add this support

RyanEwen, 2024-06-27:
There are quite a few issues / feature requests for this. It's really needed

gary-wlg, 2024-08-28:
Has there been any movement on this feature? I, as many others have said find it very frustrating to have to move through hundreds of files on each printer to start a job.  Looks like this was opened almost 2 years ago and was being evaluated.

CCS86, 2024-08-28:
@sodachen22 Can you give us an update please?

jryan15, 2024-12-30:
+1 sad this is not available on the flagship 3d printer

oO-Markus-Oo, 2025-02-11:
+1 for me. Now where Bambu plans to deactivate MQTT support this becomes even more important. A folder structure would allow a easy start of prints via Handy and / or directly from the printers without the need to slice it everytime.

Sure you can do this also without folders - issue is that you won’t find the gcode you like to print 🤪

CCS86, 2025-02-11:
Unfortunately, even if they add this feature, which I really doubt at this point, when you start prints off the SD card, you give up all the notifications on Bambu Handy. So, if the printer has some sort of issue, it will not notify you. 

This is such a ridiculous gap in functionality. I have pushed their support to fix this and they won't. They act as though it's not possible for the printer itself to tell "the cloud" that a print has started, and if "the cloud" didn't initiate the print, notifications cannot be sent.

untotren, 2025-05-10:
HOW IS THIS STILL NOT IMPLEMENTED?!?! THIS IS COMPLETE INEPTITUDE.

untotren, 2025-05-10:
> Unfortunately, even if they add this feature, which I really doubt at this point, when you start prints off the SD card, you give up all the notifications on Bambu Handy. So, if the printer has some sort of issue, it will not notify you.
> 
> This is such a ridiculous gap in functionality. I have pushed their support to fix this and they won't. They act as though it's not possible for the printer itself to tell "the cloud" that a print has started, and if "the cloud" didn't initiate the print, notifications cannot be sent.
> 
> I have honestly given up hope that they still want to improve these things. Their dev team on here used to be responsive, and they have just checked out and started ignoring everything. Even things they promised to fix or look into. No replies in many months. This is a really bad choice when the market is becoming more competitive from a hardware perspective.
> 
> I will be getting a different brand for my upcoming printer because of this.

Let me know what you switch to please, I'd love to buy something from a more responsive company.