bambulab/BambuStudio #1005: Print sequence by colour

rkarlsba, 2023-01-02:
We have print by layer or by object, which is fine, but if you have, say, two blue objects and three red ones, it'd be nice to be able to print a group of objects at once, grouped by colour. This way, a way larger amount of objects can be printed in the same run. Would this be hard to add?

TheHogarthWorkshop, 2023-01-03:
doesn't it already do this?

rkarlsba, 2023-01-03:
No, it doesn't. If you print three objects where two are colour A and one is colour B and set it to print by object, it will print all by object. What I suggest, is allowing to print by layer where the colour is the same and thus allowing for more objects on the print bed, since there's no need to separate those of the same colour, as in print by layer, as much as those that will be printed separately. I can, of course, print objects of the same colour in different prints to achive the same, but if that's a long print, it'd be better to have it easily automated.

rkarlsba, 2023-01-06:
[image]
Also, the print order is wrong with print by object. Instead of printing by colour, it just chooses a random piece. This makes it flush excessively

lanewei120, 2023-03-02:
when print by object
you can change the print order by drag the object in the object-list

you can refer to this wiki https://wiki.bambulab.com/en/software/bambu-studio/sequent-print for the details

rkarlsba, 2023-03-02:
I know, but what I'm suggesting is an extension to this. As the article states, "Print-by-object can be used when we want to print object one by one." I'd like to print object of the same colour in a group and then move to the objects of another colour, so very similar to one-by-one, only that you have a group of reds and then a group of blues and so on and print each of them separate.

rkarlsba, 2023-03-02:
Also, please note that this is needed since if I use "by object" alone, the needed free area between each object reduces the total object count drastically. This would be a lot better if same-colour-objects could be grouped and printed as one.

lanewei120, 2023-03-02:
do you mean when using "by object"
the objects in the group with same color will still be printed by layer?

rkarlsba, 2023-03-02:
What I suggest is that if you have several objects of the same and different colours, grouping those objects into colour groups and printing each group by layer, will both speed up prints lightly and most importantly, make better use available space on the bed. 

Vennlig hilsen 

roy 
-- 
Roy Sigurd Karlsbakk 
(+47) 98013356 
-- 
I all pedagogikk er det essensielt at pensum presenteres intelligibelt. Det er et elementært imperativ for alle pedagoger å unngå eksessiv anvendelse av idiomer med xenotyp etymologi. I de fleste tilfeller eksisterer adekvate og relevante synonymer på norsk.

lanewei120, 2023-03-02:
thanks for the suggestion

it is a good idea, which will be discussed internally soon

lanewei120, 2023-06-05:
it is a good feature which it is in our futture plan

however not in recent versions

speleobrad, 2023-10-06:
Great suggestion - would love to see it automated, but this works too :)