bambulab/BambuStudio #1006: [ISSUE] X1C is unable to print low quality models

H4kunaM4tata, 2023-01-03:
As the titles says, if the model you are printing is low quality, forget about it, X1C won't print it.

This is how the model looks like:

[image]

This is how the printed model looks like 
[image]

I have spent countless days troubleshooting this printer to actually print high quality models no matter the STL file, it just cannot.
At this stage I have to:

1. From the internet: Check if the model has a very high quality or the printer won't print. Only download high quality/SLA models
2. My design: Anything with curves, add as many edges as possible and as consequence, a simple round adapter will be dozens of MB in size. That is the only way X1C can print as you would expect.

I mean, look at this print, how can somebody look at it and say "it is fine"?

1. flow calibration
2. pressure advance calibration
3. endless fighting with the CF rod to stop transferring resonance to the print
4. only have high quality models
5. etc and etc and etc

$1.8k 3D Printer, I shouldn't need to spend all this time to make it print or worry about the model quality.
My previous 3D printer, slice and print no matter what and you had a nice quality print.

I mean, I must be doing something terrible wrong with this printer because man, where is the printing quality I am used to have for the last 2y with my previous printer?????

Please, don't tell me I need to start using STEP instead coz this is gonna get ugly.

knbpixels, 2023-01-03:
Maybe expand on the issue as you see it...is it the layer lines?  Overall surface smoothness?  The "random" looking dents in the cheek (which look like they are in the model as well)?  And yes, STEP files do come out smoother, but based on your seemingly visceral reaction I assume they come with drawbacks.

H4kunaM4tata, 2023-01-04:
> Maybe expand on the issue as you see it...is it the layer lines?  Overall surface smoothness?  The "random" looking dents in the cheek (which look like they are in the model as well)?  And yes, STEP [...]

Overall quality is terrible, all the shining/smooth finish all mixed together, not just the cheek

I'm having to only download high quality/SLA models for X1C to print what you expect from a $2k 3D printer.

My previous 3D printer, low poly, high poly, this was never an issue just slice and print.

There's nothing else on the slicer side that will improve this other than print using a matte filament which hides all the imperfections.

But that is not solving the problem, is it???

Flow, PA and temp have been calibrated, I must be missing something on the slicer.

knbpixels, 2023-01-04:
I'm not seeing any issue in your print that don't exist in your model.  Is your complaint that the printer is too accurate and because of that flaws appear?  I don't see different finish effects in your print like you're talking about, I see roughness of the model that cause it to not look smooth, and really fat layer lines.  If "overall quality is terrible". I just don't see that in the print.   Maybe show some other print/model examples

RobinWhitfield, 2023-01-04:
This is certainly an interesting topic, and something which has come up a few times.

#228 is probably most relevant, but I do think this has been improved with the latest slicer defaults and firmware improvements.

For the specific model that you're printing, I understand what you're geting at, but it _does_ seem to have been printed pretty faithfully. The fact the model is quite low poly is definitely some of the issue here.

Adjusting the resolution setting in the slicer would likely improve things, generating more arcs rather than straight lines

If you're having issues with varying surface finish, turn off the "slow down for overhangs" option. This is fine for functional prints, but for anything aesthetic it can cause a lot of problems.

In short I think there's probably more work to be done with #228, and the slicer could definitely do with more profile defaults to optimise for things other than pure speed.

barrettj, 2023-01-05:
I agree that it looks like it faithfully reproduced that (possibly scanned) model.  You can straight up see the triangles that make up the model in a lot of places, so it's actually surprising how smooth the printed model ends up.

Maybe take the STL into meshmixer or something and smooth it?

H4kunaM4tata, 2023-01-08:
@RobinWhitfield I have replied to your ticket, this problem is far beyond we can solve ourselves.

H4kunaM4tata, 2023-01-10:
This ringing is far worse than I expected.
Find a SLA or high quality poly only, X1C cannot print low poly models, no matter how good other printer might print the very same model.

At this stage there is no real solution from BambuLab and since BambuStudio latest release outer/inner walls acceleration was reduced from 10k to 5k to improve quality which is what the community have been doing to print somewhat high quality prints, tells you that the problem is a lot bigger.

I am closing this

closed 2023-01-10 by H4kunaM4tata