bambulab/BambuStudio #1007: [Bug] Spaghetti detection needs work

TheCableGuy99, 2023-01-03:
**Describe the bug**
Spaghetti detection on the least sensitive setting (low) is giving false positives when there's absolutely nothing to suggest a failure. 

Please see the attached screenshots one which shows the print had been paused due to spaghetti detection and the other with the current state of the print (paused by the machine). There's just nothing gone wrong so I don't know why it failed.

I've had to put both my printers on the lowest setting because it keeps happening occasionally but as this shows even the lowest setting has issues.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Print something with spaghetti detection on

**Expected behavior**
To only pause when an actually failed print occurs.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - Windows 11 Pro
 - Bambu Studio 1.4.1.52

c2h5oh, 2023-01-03:
Test something for me: please put something on top of the printer to block the light - a piece of cardboard, cloth, doesn't matter as long as it makes the light inside the printer the source of majority of light in the printer.

I started having the problem you've described after adding led strips on top of the printer and in my case it goes away if I turn off the lights as it prints first layers. I see the interior of your printer is fairly bright, so it could be the same thing.

TheCableGuy99, 2023-01-03:
Hey, Yeah I've got LED panels sat on top of the glass. It's going to take a while before I know because it doesn't happen every print, it's just now and again. So I can't say conclusively unless I turn the light off every print and it doesn't happen for about a week.

SaltWei, 2023-01-04:
This has no relationship with Slicer and maybe official support can't be got timely here.
For firmware and printer issue, please move to https://forum.bambulab.com/ for better support. Thanks!