bambulab/BambuStudio #1012: Manage on-printer files through Bambu Studio

Fl0ken, 2023-01-04:
**Is your feature request related to a problem? Please describe.**
Uploading the sliced files and being able to select as well as delete them from the touchscreen is great. A great next step would be to start as well as manage the files from Bambu Studio itself (similar to the G-Code Files in Mainsail/Fluidd).

**Describe the solution you'd like**
Add an additional tab in the printer menu (like Media) where we can see all stored gcode files and either start or delete them. 

**Additional context**
Could be limited to X1s. Would be great to have it on P1Ps too. 
Additionally maybe a check (e.g. "Please check if the printbed is clear") before printing should be implemented to avoid any crashes.

adonno, 2023-01-16:
that would be awesome

bambu123, 2023-03-02:
Yes, we are considering that.