bambulab/BambuStudio #1013: Cant use the full bed cause it "needs" that section for the cutter? lower the bed....

Benderrodrigu3z, 2023-01-04:
its honestly stupid that it needs that bed section for filament cutting, it could just lower the bed cut it then raise the bed back up, this would only become an issue when the bed is at the bottom, cut it out then

Benderrodrigu3z, 2023-01-04:
also where is the default use all bed profile with key override?

Benderrodrigu3z, 2023-01-04:
so like dropping the bed 10mm but think about it while its purging drop the bed, after it cuts while its purging raise the bed
0 time loss

Benderrodrigu3z, 2023-01-04:
lower the bed then move it to the cutter, thats the only time lost, please give us this as an option

Sirwill1968, 2023-01-04:
Feature: Lower enough to cut without hitting model. Go purge.  While purging bring back up to print height the print bed saving time.  Make it so this only takes place if object placed in exclusion zone.  Then exclusion zone only truly applies for top XXmm of the Z height.

Benderrodrigu3z, 2023-01-04:
doing this you would still not be able to use the entire bed because if it moves there it cuts it but this would make it so you could cut the white section in half as half of it doesn't hit the cutter on movements

RobinWhitfield, 2023-01-22:
Whether you lower the bed or not, the cutter lever would still be hit by the toolhead.

The model being printed is always lower than the nozzle anyway.

SaltWei, 2023-03-02:
It's impossible to solve the problem by lower bed. This is mechanical limitation. Please see this wiki.
https://wiki.bambulab.com/en/knowledge-sharing/print-volume-limitations.

closed 2023-03-02 by SaltWei