bambulab/BambuStudio #1014: [BUG] [MacOS] Can't scroll down the options to save files, plates, slices...

bolsoncerrado, 2023-01-04:
**Describe the bug**
https://m.facebook.com/groups/bambulabofficialusers/permalink/930213347922685/ 

**To Reproduce**
MacStudio M1 running latest MacOS Ventura update and latest BLStudio software. 

Just trying to save file gcode or do anything other than Print Plate.... 

**Expected behavior**
The options to be usable.

lanewei120, 2023-01-05:
what is your studio version number?
1.4.0 or 1.4.1?

we have fixed a similiar issue in 1.4.1
if you are using 1.4.0, please have a try for 1.4.1 here: https://github.com/bambulab/BambuStudio/releases/tag/v01.04.01.52

bolsoncerrado, 2023-01-05:
1.4.0.18

So I have to update manually? OK I'll do and report back.

bolsoncerrado, 2023-01-05:
OK "Check for updates" was not detecting it so I manually installed it and yes, problem is gone.

It only  has a visual glitch with the mouseover you may want to fix at a later moment...

https://user-images.githubusercontent.com/10992914/210732842-e1732220-b6fc-461e-a844-d35982356816.mp4

Benderrodrigu3z, 2023-01-05:
you don't save the profiles there, see below pics
@bolsoncerrado 
[image]
[image]
[image]

bolsoncerrado, 2023-01-05:
> you don't save the profiles there, see below pics @bolsoncerrado [image] [image] [image]

So what? It was solved already on 1.4.1

Nothing to do with unsaved profiles...

Benderrodrigu3z, 2023-01-07:
okay so I think what your asking for is to save the stl file with all this information correct? if yes that means that you need to do a File->Export->Generic 3MF

bolsoncerrado, 2023-01-07:
No, my issue was already resolved (GUI GLTICH)

closed 2023-03-02 by SaltWei