bambulab/BambuStudio #1015: Inconsistent seam placement

The760, 2023-01-04:
A printed parts seam placement is completely different compared to the sliced g-code preview. In the pictured example, you can see that the seam is aligned in the preview. But on the printed part, the seam is completely inconsistent in comparison. I tried different seam positions, and even tried seam painting, but it results in the same inconsistencies.

[image]

The model is work related so I cannot share it. However, these results should be easily reproducable on any model.

I have seen a few other people say they have experienced the same bug, so I hope other users can chime in to support my claim with more picture examples.

 - OS: Windows 10
 - Version 1.4.1

NairoDorian, 2023-01-05:
I have noticed the same problem with my seam, but more importantly, I find the seam extremely visible on my parts, this looks like a slicer bug, I did not notice this problem with the first version of the slicer & firmware back when we Kickstarter received our units in the first batchs
I might be wrong but maybe this bug as been introduced since an update or a feature change, could it be the gcode between each layer? Is it related to the Timelapse feature? Could be so many things

SaltWei, 2023-01-05:
Did you enable timelaps?

SaltWei, 2023-01-05:
> I have noticed the same problem with my seam, but more importantly, I find the seam extremely visible on my parts, this looks like a slicer bug

Could you share you model for us to reproduce?  By the way, there has one known firmware bug which may cause seam more visible when enable timelaps.

The760, 2023-01-05:
> Did you enable timelaps?

Nope, I haven't used it at all yet since I got my X1 Carbon. I disable it before the start of each print, and I don't have a MicroSD in the machine either,

The760, 2023-01-05:
Okay, I've figured out the cause of this bug. **It's Arachne.**

I sliced the same model with Classic wall generator instead of Arachne, and the seam is correct. I tried a different model as a sanity check, and the seam is also correct on that one.

For Arachne, I didn't change anything. I kept the stock settings. I don't know if it's an implementation bug, or if one of the settings is causing it. But I've narrowed it down to something with Arachne being the root cause.

If the team could look into this, it would be much appreciated!

QingZhangBambu, 2023-01-09:
> Okay, I've figured out the cause of this bug. **It's Arachne.**
>
> I sliced the same model with Classic wall generator instead of Arachne, and the seam is correct. I tried a different model as a [...]

ok, I will check it. Your idea is helpful.

QingZhangBambu, 2023-01-17:
> A printed parts seam placement is completely different compared to the sliced g-code preview. In the pictured example, you can see that the seam is aligned in the preview. But on the printed part, [...]

could you provide the model? I try some, but the seam position is right.

QingZhangBambu, 2023-06-20:
we can not reproduce that, could you provide more information?

QingZhangBambu, 2023-08-08:
closed for no response

closed 2023-08-08 by QingZhangBambu