bambulab/BambuStudio #1016: Scrolling can cause values in input with arrows to change

Hologos, 2023-01-05:
**Describe the bug**
When scrolling if the mouse pointer stops above an input with arrows, next scrolling changes the value instead of scrolling.

This happens when you are scrolling and need to re-adjust the finger to scroll some more.

**3mf File for This Bug**
No need

**To Reproduce**
Steps to reproduce the behavior:
1. Place a mouse pointer so when you scroll you hover over inputs.
2. Begin the scroll.
3. Stop scrolling over an input with arrows.
4. Scroll some more.

**Expected behavior**
The scrolling continues, no value is changed.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Version: 13.1

bambu123, 2023-01-05:
Yes, the scrolling should be disabled in the SpinCtrls here.

SaltWei, 2023-03-02:
Available in v1.5.0 public beta

closed 2023-03-02 by SaltWei