bambulab/BambuStudio #1018: Rectilinear Grid Pattern for Interface Pattern.

wirlybird, 2023-01-06:
When the interface extrusions all run in the same direction for multiple layers they tend to start to curl at the edges.
If they curl enough the nozzle can catch on them.
Having the layers run perpendicular helps to prevent this.

SaltWei, 2023-01-09:
Could you give more explanation about this by some pictures？

tangketan, 2023-03-21:
Hi @wirlybird , sorry that I can't find any picture attached. But there is another issue about this. See #1221 .
I give an example of "Grid" interface pattern. Is it what you want?

wirlybird, 2023-03-21:
From my experience I think rectilinear would be best rather than “Grid” pattern.

tangketan, 2023-03-21:
Do you mean an alternative perpendicular linear pattern? Eg. 1st layer is all horizontal lines, then 2nd layer is all vertical lines?

wirlybird, 2023-03-21:
Basic rectilinear pattern.  See pictures.
One layer is at a 45 to the X,Y axis, picture 1, then the next layer of the dense support is 90 degrees from the previous layer, picture 2.

In the example I have dense support set to 100% so it is solid.  Works best for me.

tangketan, 2023-03-21:
I still can't see your pictures... But I think I've got your point. Basically you want the existing rectilinear pattern, but with perpendicular angles at neighboring layers, right?
[image]

wirlybird, 2023-03-21:
Yes, I think that is it.
My pictures are attachments to my email.

 

 <https://user-images.githubusercontent.com/13730459/226530191-ea1efd39-04bb-4f4c-a9e2-f6e8d4af43f6.gif> 20230321-140721

wirlybird, 2023-03-21:
Here it is again.  Pictures in email.

 

Basic rectilinear pattern.  See pictures.
One layer is at a 45 to the X,Y axis, picture 1, then the next layer of the dense support is 90 degrees from the previous layer, picture 2.

In the example I have dense support set to 100% so it is solid.  Works best for me.

wirlybird, 2023-03-24:
Here is a good example of needing to be able to set the “angle” of the infill pattern.

In this case, on two sides of the print, the dense infill is at 90 degrees to the layer above but on the other two sides, red arrow, the last layer of dense infill runs parallel to the layer above.  This can cause the layer to bond more aggressively to the support than when the support layer is not parallel to the layer above it.

tangketan, 2023-04-07:
Email reply of github doesn't seem to work correctly. I didn't see any picture in any of your email replies.
Anyway, I've added the new interface pattern and pattern angle.

psy0rz, 2023-04-25:
check out https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65 

can probably be closed now.

closed 2023-04-26 by lanewei120