bambulab/BambuStudio #1019: Add Patterns For Internal Solid Infill Same As Top/Bottom Surface Patterns.

wirlybird, 2023-01-06:
Please add the ability to select the internal solid infill pattern just like we can for top & bottom surfaces.
Maybe it is just a check box for internal solid infill pattern that says, "same as top surface pattern" or "same as bottom surface pattern".
Or it has its own drop-down selection.

In some situations having the internal solid infill be a concentric type of infill (all extrusions running parallel) creates a weak structure.  Especially if the area is narrow.

SaltWei, 2023-01-06:
> In some situations having the internal solid infill be a concentric type of infill (all extrusions running parallel) creates a weak structure. Especially if the area is narrow.

This situation can be handled by disable “Detect narrow solid infill”
[image]

wirlybird, 2023-01-06:
Thank you.  That does work for the case I presented.

However, I would still like to be able to pick the infill pattern for internal solid infill same as we can for top/bottom surfaces.

 

Thank you again for considering this and your assistance.

 

 <https://user-images.githubusercontent.com/110660407/210975696-d72edbf9-9aad-42d6-8f65-a1f91f4968c3.png> image

SaltWei, 2023-01-09:
> However, I would still like to be able to pick the infill pattern for internal solid infill same as we can for top/bottom surfaces.

More settings can give the user more freedom, but also make the setting page to be complex and not easy for newbie.

Could you give some cases to show this setting is necessary (otherwise, the printing is not perfect)?

wirlybird, 2023-01-09:
HI,

 

I think there are infill types that would be beneficial.  I also think Concentric is not one of them.  I think Concentric is more of an Ornamental type as it is a low strength pattern.

 

Maybe if we had a drop-down box like we do for Top/Bottom layers with some choices that would be good.

Or, we had check boxes so solid infill would be the same as what is used for top/bottom layers.

 

In other slicer when I pick top/bottom layers I pick an infill type and how many layers for each top & bottom.

 

I think that the detect narrow internal solid infill should use a rectilinear grid style pattern rather than concentric.  Grid is much stronger and there is a much better chance the extrusions will not be running para

The intended purpose of this setting is not practical.  Concentric is not a good infill choice and using it really does not save any time as opposed to rectilinear grid since the affected areas will be small.

So maybe internal solid infill follows the pattern selected by top/bottom unless it falls into the criteria for “detect narrow internal solid infill” then it reverts to a grid type infill.

SaltWei, 2023-06-20:
Will be included in next release version

SaltWei, 2023-07-11:
[image]
Included in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.53
Thank you very much for suggestions!

closed 2023-07-11 by SaltWei