bambulab/BambuStudio #1022: [BUG] Printer does 5 mesh check instead of the 36 mesh checks leaving the first layer calibration broken

H4kunaM4tata, 2023-01-07:
To make things clear, I was using `G29.1 Z-0.04` for PEI before BL made it available.
I don't believe PEI is the problem.

These gold lines aren't light, they are the sheet seen through the first layer
[image]

I have noticed that some of my prints have a so so first layer finish due to gaps between the first lines why others are fine and I believe I found why.

How to replicate:
Started another print after I had to cancel this same print due to another bug with the cooling logic, and do the usual, select only bed calibration and time-lapse.
The printer then does 5 mesh check, one in the middle and all 4 corners (?????)

My previous printer was 49 mesh check or 7x7, X1C is only 36 or 6x6 and now it is down to 5?? I mean....

If I select the printer to do bed calibration, I expected it do perform the full bed calibration.
I don't wanna it using some value from previous print for whatever reason.
Removing the model from the bed, moving the sheet is enough to mess the bed calibration.

Solution: If bed calibration is selected, perform a full bed calibration or the 36 mesh check and not 5 mesh check for whatever reason.
Commodity or to save time isn't excuse to perform 1/4 of the bed calibration process when the quality delivered is bad.

This has to be fixed asap.

AddChickpeas, 2023-01-10:
I've also had issues with the bed leveling not performing the full sequence. 

On the P1P, it normally does a 5x5 mesh, but sometimes it just does a 5 point probe (corners and center). Haven't figured out what causes it/how to force it to do a full 5x5 probe. 

My guess, is that it does the corner probe and, if it's similar to the previous mesh, it skips it or something like that, but I'm just guessing. 

I was doing some large flat prints and having some leveling issues and it was very frustrating to restart a print and for it to only probe the corners for some reason.

H4kunaM4tata, 2023-01-10:
@AddChickpeas X1C appears to do that when you hit "reprint".
I confess I have no noticed this before because I hit start usually and walk away. I never ever hit reprint.
This print tho, as I noticed some inconsistencies, I was looking closer and noticed this behavior.

SaltWei, 2023-01-11:
> Solution: If bed calibration is selected, perform a full bed calibration or the 36 mesh check and not 5 mesh check for whatever reason. Commodity or to save time isn't excuse to perform 1/4 of the bed calibration process when the quality delivered is bad.
> 
> This has to be fixed asap.

Thank you very much for suggestions!
But this need to be handled by printer and maybe official support can't be got timely here(For slicer).

I have transfered this ticket to printer team.

For future issue and suggestion about firmware and printer, moving to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) can get better support. Thanks!

SaltWei, 2023-01-11:
Response from printer team:
5 mesh check is just used to check whether the previous 5x5 mesh level data is still valid.If not, the printer will do 5x5 mesh probe again.

H4kunaM4tata, 2023-01-11:
> Response from printer team: 5 mesh check is just used to check whether the previous 5x5 mesh level data is still valid. If not, the printer will do 5x5 mesh probe again.
> 
> If you don't want the 5 mesh check, and always want 5x5 mesh, you can change the start gcode and delete the 'A' in "G29 A ... " line [image]

Yes please, that will do. Thank you for that :)

Side note, X1C is 6x6 and P1P appears to be 5x5

closed 2023-01-11 by H4kunaM4tata

SaltWei, 2023-01-11:
> X1C is 6x6 and P1P appears to be 5x5

Thanks for remind