bambulab/BambuStudio #1025: Add printer calibration features

IkarusX1978, 2023-01-07:
Please add Printer calibration features as seen in SuperSlicer with the well known models for Bed/Extruder leveling, Filament Flow, Filament Temperature, Extruder Retraction, Bridge Flow, Ironing Pattern, Final Calibration Cube/Cat Test.

FreeManFreeThinker, 2023-01-08:
Filament temp, flow and retraction all would be extremely nice to have. I'm having a hard time with ninja flex and having these options to easily find out what works best for this TPU would be awesome.

Merlwynd, 2023-01-08:
> Filament temp, flow and retraction all would be extremely nice to have. I'm having a hard time with ninja flex and having these options to easily find out what works best for this TPU would be awesome.

So...I don't understand the comment. All the items you mention....filament temp, flow, retraction, retraction speed, etc. are options you can change in Bambu Studio on a per filament basis....

FreeManFreeThinker, 2023-01-09:
> So...I don't understand the comment. All the items you mention....filament temp, flow, retraction, retraction speed, etc. are options you can change in Bambu Studio on a per filament basis....

It appears that the request is for a feature in Bambu Studio that allows users to print specific models in order to fine-tune the calibration for non-Bambu filaments. 

On the website provided, there are a variety of models available for printing that can be used to adjust certain settings on the printer. For example, a 4-pole retraction test can be printed, and the gcode will vary the retraction throughout the print to help determine the optimal setting. 
[](https://teachingtechyt.github.io/calibration.html#intro
)
Although Bambu typically does a good job of calibrating itself, it may be useful to have filament temperature, flow, and retraction tests available for fine-tuning filament settings.

Merlwynd, 2023-01-09:
I see...a matter of convenience then as these models all available on the web (e.g. Prusa hollow calibration cube for flow). Might be interested in BambuLab Studio Softfever fork...as that has a couple of calibration models built into it (just flow and linear PA)...