bambulab/BambuStudio #1027: v1.4.1 Slim Tree Supports - No way to control the Slim Tree Support Brim Size

UncleJessy, 2023-01-08:
**Describe the bug**
When using the new Slim Tree Supports - There is no way to adjust the Brim Size that is generated with the new Slim Tree Supports. The issue is that my Model/Supports fit on the buildplate but the Brim causes the print to go over the edge of the buildplate. If I could make the brims smaller it would fit properly for printing. 

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Add any model that Requires supports 
2. Generate the new Slim Tree Supports (Manual or Auto)
3. It will generate Brims for each of the tree supports
4. There doesn't appear to be a way to set how many Brims are generated per support tree. 

**Expected behavior**
A setting to control that or use the existing Brim controls under Others menu

**Screenshots**
https://imgur.com/a/zlzlZyp

**Desktop (please complete the following information):**
 - OS: Mac OS 12.1

tangketan, 2023-01-09:
Hi @UncleJessy , We plan to use the build plate to trim the support volume, which guarenttees no support brim goes outside the bed. Is that OK for you?

UncleJessy, 2023-01-09:
> Hi @UncleJessy , We plan to use the build plate to trim the support volume, which guarantees no support brim goes outside the bed. Is that OK for you?

That would be fantastic for any of the support options. I'm continuing to push the build volume limits. Something like that would be fantastic.

loraderon, 2023-02-09:
> We plan to use the build plate to trim the support volume, which guarenttees no support brim goes outside the bed. Is that OK for you?

Make sure that you include "the forbidden spot" when trimming the support brims.

[image]

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei