bambulab/BambuStudio #1029: Access Violation when slicing colored model

SwagDad99, 2023-01-08:
**Describe the bug**
After spending hours coloring a detailed model, it  gets an 'Access Violation' when slicing.  In the thumbnail in the upper right corner of the preview window while slicing, it gets the error when it gets down to the first colored area of the model.  The progress text says "35% Generating Infill Toolpaths"

**3mf File for This Bug**
The file slices fine before coloring.  The STL is a paid file, and I am unable to share it without permission.

**To Reproduce**
Steps to reproduce the behavior:
1. Spend hours coloring a model in great detail.  (Perhaps generating too much detailed color?)
2. Click on 'Slice'
3. See error 'Access Violation'

**Expected behavior**
Successful Slice and ability to print.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version [e.g. 22]

SaltWei, 2023-01-11:
Which windows version are you using?

Could you try following binary https://www.dropbox.com/s/cm3a4v0x9vc5ixp/Bambu_Studio_win-v01.04.01.52-20221226111441.zip?dl=0
which includes symbol table for generating slicing log.
and try to slice again.

when it crashes
please help to share the log file and the crash log file under C:\Users\user.name\AppData\Roaming\BambuStudio\log
such as followings:
image

SwagDad99, 2023-01-12:
I am using Windows:
Edition    Windows 11 Home
Version    22H2
Installed on    ‎5/‎7/‎2022
OS build    22616.1
Experience    Windows Feature Experience Pack 1000.22632.1000.0

It still fails with the same error.  Attached are the logs.

Don

From: SaltWei 
Sent: Tuesday, January 10, 2023 7:39 PM
To: bambulab/BambuStudio 
Cc: SwagDad99 ; Author 
Subject: Re: [bambulab/BambuStudio] Access Violation when slicing colored model (Issue #1029)

Which windows version are you using?

Could you try following binary https://www.dropbox.com/s/cm3a4v0x9vc5ixp/Bambu_Studio_win-v01.04.01.52-20221226111441.zip?dl=0
which includes symbol table for generating slicing log.
and try to slice again.

when it crashes
please help to share the log file and the crash log file under C:\Users\user.name\AppData\Roaming\BambuStudio\log
such as followings:
image

SaltWei, 2023-01-12:
> Attached are the logs.

Sorry, I can't see any attachment. May be limited by github.
Could you please send the log to my email: salt.wei@bambulab.com ?

SwagDad99, 2023-01-13:
I just sent them.

SaltWei, 2023-01-13:
Not sure what's the problem, I have refreshed email for many times but still get nothing except message from github. Have contacted with IT guys wo check whether has networt issue
[image]

99berend99, 2023-01-13:
I got the same error. Also a colored model.

Mine got stuck on 15%, generating walls.

[image]

SaltWei, 2023-01-16:
> I got the same error. Also a colored model.
> 
> Mine got stuck on 15%, generating walls.

Thank you very much for reporting this issue and sharing the 3MF file!

For this 3mf file, this is an known issue caused by fuzzy skin and we have fixed in internal version.

The fix will be included in the next user version, which will come soon!

SwagDad99, 2023-01-16:
So, my model does not have fuzzy skin.  That means my issue is separate, and not fixed in the next release?

SaltWei, 2023-01-16:
> So, my model does not have fuzzy skin. That means my issue is separate, and not fixed in the next release?

There are no abnormal log in the file you shared.
Is there log file which name begin with "crash_" after using the binary I shared to slice?

SaltWei, 2023-01-16:
It would be very convenient to debug the issue if had one 3mf shared which has no copyright.

SwagDad99, 2023-01-16:
I am sorry, but the only two 3mf files that I have had this issue with are purchased files.  You could contact the Patreon and see if he will allow me to share it with you. https://www.myminifactory.com/users/dansopala
Tell him that it is the Coal Miner Ant.

Don 

From: SaltWei 
Sent: Sunday, January 15, 2023 10:54 PM
To: bambulab/BambuStudio 
Cc: SwagDad99 ; Author 
Subject: Re: [bambulab/BambuStudio] Access Violation when slicing colored model (Issue #1029)

It would be very convenient to debug the issue if had one 3mf shared which has no copyright.

SwagDad99, 2023-01-17:
I now have the error with a file I can share...!  For some reason the printer/nozzle type doesn't want to stay X1C .04, and the layer height should be 0.20 standard.
[Print-in-Place - Small-Colored.zip](https://github.com/bambulab/BambuStudio/files/10432213/Print-in-Place.-.Small-Colored.zip)

SaltWei, 2023-01-17:
> I now have the error with a file I can share...! For some reason the printer/nozzle type doesn't want to stay X1C .04, and the layer height should be 0.20 standard.

Thank you very much and now we can reproduce this issue.

SaltWei, 2023-01-17:
@zhimin-zeng-bambulab Please check this array which should be 8x8,but it's 4x4 now
[image]
[image]

SwagDad99, 2023-01-17:
I went to flushing volumes, and yes, there are only 4 on the left, and 4 on the top...  How do I fix that?

Don

From: SaltWei 
Sent: Tuesday, January 17, 2023 12:35 AM
To: bambulab/BambuStudio 
Cc: SwagDad99 ; Author 
Subject: Re: [bambulab/BambuStudio] Access Violation when slicing colored model (Issue #1029)

@zhimin-zeng-bambulab Please check this array which should be 8x8,but it's 4x4 now

SaltWei, 2023-01-17:
Add one filament and then delete it. The array will be updated to be 8x8. We think this is one issue caused by UI and will fix it

SwagDad99, 2023-01-18:
It worked!  Thank you!

Don

From: SaltWei 
Sent: Tuesday, January 17, 2023 3:33 PM
To: bambulab/BambuStudio 
Cc: SwagDad99 ; Author 
Subject: Re: [bambulab/BambuStudio] Access Violation when slicing colored model (Issue #1029)

Add one filament and then delete it. The array will be updated to be 8x8. We think this is one issue caused by UI and will fix it

closed 2023-03-30 by zhimin-zeng-bambulab