bambulab/BambuStudio #1030: Broken dark/light theme when mac is woken up

Hologos, 2023-01-08:
**Describe the bug**
I was working in BambuStudio at night and sleeped my Macbook. When I woke the mac up, the theme is broken and I dont see a way to fix it (other that restart the program).

**3mf File for This Bug**
--

**To Reproduce**
Steps to reproduce the behavior:
1. Set dark mode to automatic (in macOS)
2. Open it at night.
3. Sleep the mac.
4. Wake it "non-night" time.

**Expected behavior**
It can apply correct theme.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: macOS
- Version: 13.1
- Bambu Studio v1.4.1.52

SaltWei, 2023-01-11:
@bambu123 Please help to check and follow this ticket. Thanks!

pdelvo, 2023-01-23:
Mine looked the same at some point on Windows. But I was not able to reproduce it so I dont know what caused it. Enabling dark mode in the settings fixed it though without restarting

VACIndustries, 2023-02-19:
Also seeing this on Windows when I remote desktop into the machine.  You have to check the box again to "turn on" dark mode again.

bambu123, 2023-02-20:
Some log was added for this issue. I'm sorry for not getting any more progress.

SaltWei, 2023-03-02:
More logs have been added into v1.5.0 beta version to help to reproduce and analyse.
Please let us known when it reproduce again and we will reopen this issue.

closed 2023-03-02 by SaltWei