bambulab/BambuStudio #1036: Send to multiple printers

TheHogarthWorkshop, 2023-01-09:
**Describe the solution you'd like**``

Please allow sending files to multiple printers from the 'Send' dialog.

Simply removing the 'Printer' dropdown and having a checkbox of All, Printer1, Printer2....etc would work. 

This dialog should also have a universal print 'sd card' folder structure that it saves the files into on all printers. 

This is actually a massive issue for me, with the kind of orders produce, and would massively improve my workflow.

TheCableGuy99, 2023-01-10:
+1 for this, I was just about to post this as well :)

rhys100, 2023-02-19:
+ 1 Need to send the same print to 10 printers at once and its very slow and tedious one by one

NitsuaFox, 2023-08-02:
> + 1 Need to send the same print to 10 printers at once and its very slow and tedious one by one

any work around on this ?

rhys100, 2023-08-03:
@lanewei120 
are these machines using AMS? yes

and if send the same print to multiple printers
do you need to do AMS mapping for each printer? not usually. usually it would be PETG black / one colour and i will load all the printers and set the material/colour on each printer, then return to computer but i think you could do it in the one popup if there was multiple colours as well. select printers that you want, ams mappings would load and you select them all on the one screen then click print all

TheCableGuy99, 2023-08-03:
> Hello @NitsuaFox @HOLLY-DION @rhys100 @TheCableGuy99
> 
> are these machines using AMS? and if send the same print to multiple printers do you need to do AMS mapping for each printer?
> 
> thanks

Hi, Yes each machine has more than 1 AMS. However I would personally ensure that each machine has the same filament loaded in the same slots. 

But I have 3 machines and 2 of them have 3 AMS and 1 machine has 2 AMS.

So as long as I can send to all 3 machines (and if I wasn't using more than 8 filaments) it doesn't give an error because one machine has 1 less AMS which wouldn't be being used anyway.

Thanks.

NitsuaFox, 2023-08-15:
Same as TheCableGuy99's comments. Really at the moment it's just a manual task 'SENDING' to 1 printer, then waiting, then selecting the next printer and selecting SEND again... To top it off the count down to close the window is absolute nonsense, at the very least remove that!  

Seriously though, just change the dropdown to checkbox(s)  on the 'Send to Printer' dialog.

NitsuaFox, 2023-08-15:
Just to be clear, I just allow the AMS mapping to take place automatically based on the colours used, I don't have the time to change device on slicer, sync and reassign colours,

NitsuaFox, 2023-10-16:
Guys are we any closer to resolving this very simple GUI issue.

The count down from 3 seconds to 0 to close the window when sending a job to the printer is so annoying. Please remove it, it has no reason for existence. 

Also, if possible as OP. Please instead of the drop down to select printer, have a check box, so I can send to multiple printers.