bambulab/BambuStudio #1038: VPN Load Crashing issue

Benderrodrigu3z, 2023-01-09:
If my VPN is on being Windscribe and I open 01.04.01.52 it crashes on load and I have to force quit even a new installation does this.
however 01.03.01.01 doesn't do this

Which tells me that 01.04.01.52 trying to check something on the internet JUST to open every time.
Why not test once if it fails open the program and tell the user an error?

lanewei120, 2023-01-11:
@Benderrodrigu3z 
which windows version are you using?

could you try following binary https://www.dropbox.com/s/cm3a4v0x9vc5ixp/Bambu_Studio_win-v01.04.01.52-20221226111441.zip?dl=0
which includes symbol table
and try again

when it crashes
please help to get the log file and the crash log file under C:\Users\user.name\AppData\Roaming\BambuStudio\log
such as followings:
[image]

Benderrodrigu3z, 2023-01-11:
I never said Windows, this is on Mac 10.15.7Sincerely,Shane FSent from my iPhone13

lanewei120, 2023-01-11:
@Benderrodrigu3z 
Hello

for mac system
there should be a crash report

could you help to send the crash report to us?

Benderrodrigu3z, 2023-01-11:
Well when I turn off the vpn it loads, not near my laptop I’ll do it in a few daysSincerely,Shane FSent from my iPhone13

Benderrodrigu3z, 2023-01-11:
[debug_network_Mon_Jan_09_13_52_31.log](https://github.com/bambulab/BambuStudio/files/10396528/debug_network_Mon_Jan_09_13_52_31.log)
[debug_network_Mon_Jan_09_13_55_54.log](https://github.com/bambulab/BambuStudio/files/10396529/debug_network_Mon_Jan_09_13_55_54.log)
[debug_network_Mon_Jan_09_11_41_39.log](https://github.com/bambulab/BambuStudio/files/10396530/debug_network_Mon_Jan_09_11_41_39.log)
[debug_network_Mon_Jan_09_11_41_34.log](https://github.com/bambulab/BambuStudio/files/10396531/debug_network_Mon_Jan_09_11_41_34.log)
[debug_network_Mon_Jan_09_11_16_36.log](https://github.com/bambulab/BambuStudio/files/10396532/debug_network_Mon_Jan_09_11_16_36.log)
[debug_network_Mon_Jan_09_10_28_21.log](https://github.com/bambulab/BambuStudio/files/10396533/debug_network_Mon_Jan_09_10_28_21.log)
[debug_network_Mon_Jan_09_10_25_07.log](https://github.com/bambulab/BambuStudio/files/10396534/debug_network_Mon_Jan_09_10_25_07.log)
[debug_network_Mon_Jan_09_10_23_44.log](https://github.com/bambulab/BambuStudio/files/10396535/debug_network_Mon_Jan_09_10_23_44.log)
[debug_network_Mon_Jan_09_10_22_41.log](https://github.com/bambulab/BambuStudio/files/10396536/debug_network_Mon_Jan_09_10_22_41.log)
[debug_network_Mon_Jan_09_10_21_39.log](https://github.com/bambulab/BambuStudio/files/10396537/debug_network_Mon_Jan_09_10_21_39.log)
[debug_network_Sat_Jan_07_08_58_47.log](https://github.com/bambulab/BambuStudio/files/10396538/debug_network_Sat_Jan_07_08_58_47.log)

lanewei120, 2023-01-13:
@Benderrodrigu3z 

oops, not these logs, but the crash report
you can find it from 'console'

just following this guide: https://support.apple.com/en-in/guide/console/cnsl664be99a/1.1/mac/10.15

lanewei120, 2024-01-16:
@Benderrodrigu3z 
do you stlll has this issue?

Benderrodrigu3z, 2024-01-20:
> Not sure, but I can test it tonight, I don’t print a lot.Sincerely,Shane FSent from my iPhone13

Yes it works, this can be closedSincerely,Shane FSent from my iPhone13

closed 2024-01-22 by lanewei120