bambulab/BambuStudio #1040: Set/get colour RGB values

rkarlsba, 2023-01-09:
When choosing a colour with the colour picker, you can get a very precise colour match, which is fine. Hoever, if I want to set two slots to the same custom colour, there's no way to get/set the RGB value of this so even though autochange is enablied, I can only guess it won't work, since the colours have different values. Could you add RGB values (#2040ff etc) to the colour selector in BambuSlicer, please?

Benderrodrigu3z, 2023-01-09:
its there on Mac, just under a different tab
[image]

rkarlsba, 2023-01-09:
ah - there it is! My fault. Please close this.

closed 2023-01-09 by rkarlsba