bambulab/BambuStudio #1041: Infill Line Multiplier Settings

Benderrodrigu3z, 2023-01-09:
Please give us the setting for the Infilled line multiplier for using less plastic with greater strength

with a double line infilled you can make a 10% infill as strong as a 25% infill
Also please add a layer change infill amount settings to the slicer preview, so say at layer 5 we want it to be 90% infilled but layer 4 and 6 we want it to be 50 like the rest of the model, to do that we need to change it at layer 5 then change it back on layer 6 but we don't have this an an option

Reason:
https://www.youtube.com/watch?v=yskFuq8VjtA
He talks about it at marker 2:27

TeddybearCup, 2023-02-22:
This would be such a great addition. I've had several cases where I want strong but sparse infill in the past and infill multiplier really helped out.
Infill width could be a decent substitute, but only to a point.

jwilliams1976, 2023-07-01:
+1 on this feature request.

I loved using Honeycomb infill with no top or bottom layers on the Cura slicer for many models. Really miss that feature in this slicer.

mysteryjeans, 2023-09-29:
Its my go to setting for increasing strength, instead of increasing infill percentage. Although it uses more plastic but not as much as let say increasing it from 15% to 30%.

colorado1876, 2024-07-30:
+1. Please add an infill line multiplier. This is also a great option for cosmetic designs where an infill pattern is desired for appearance, but one can choose the infill size by selecting density, but also maintain structural integrity of the matter by thickening it.

As OrcaSlicer is based on Prusa Slicer, it is hard to believe this feature has been omitted, particularly as it does not seem to be difficult to implement.

RF47, 2025-12-25:
Better implementation in case you are interested:
https://github.com/OrcaSlicer/OrcaSlicer/pull/11435