bambulab/BambuStudio #1042: PREHEAT NOZZLE prior to resume a power outage print!!!

bolsoncerrado, 2023-01-09:
**Describe the bug**

I started printing 16 funnels for the 4 AMS I got. PETG print. For whatever reason power supply failed from the UPS. When I realized the nozzle was already cold.

Now here comes the fun part: upon power recovery, I'm asked to resume the print job (yeah! How cool!) just to realize the m!*!?%@# printer starts to MOVE the nozzle before heating it, not only dislodging the part it stopped onto but also the rest of parts on the path to the wiper.... ROTFL.

I MEAN, in a power failure scenario it's highly likely the printer nozzle will melt the part it stopped into and if cooled enough, they will bond temporarily.

Why in the name of John Wick do we want to move the nozzle BEFORE heating it up!!!

This is ridiculous and a bug IMHO.

**3mf File for This Bug**
N/A

**To Reproduce**
1. Start a print
2. Let it print for some height
3. Cut power and let it cool down
4. Resume the print if you dare! 

**Expected behavior**
It would be easier if nozzle preheat BEFORE starting to move at all after a power outage. 

**Desktop (please complete the following information):**
 - N/A

SaltWei, 2023-01-11:
Thank you very much!
But this has no relationship with Slicer and maybe official support can't be got timely here.

I have transfered this ticket to printer team.

For future issue about firmware and printer, please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better support. Thanks!

bolsoncerrado, 2023-01-11:
You're right, my apologies.

SaltWei, 2023-01-11:
> You're right, my apologies.

No, should be my apologies that I can't help with printer issue.

I have transfered this ticket to printer team and will respond as long as they respond.

SaltWei, 2023-01-11:
Response from printer team:
Good point and will improved in printer firmware!
Thank you very much!

bolsoncerrado, 2023-01-11:
SO AWESOME!!!

Thanks!! ;)

closed 2023-01-11 by bolsoncerrado