bambulab/BambuStudio #1043: Wider adaptive outer walls to help with overhangs

Quick-Flash, 2023-01-10:
**Is your feature request related to a problem? Please describe.**
Lately I have been printing with an extra wide outer wall. The reason being that wider nozzles handle overhang better. I've been able to see improvements to my prints that are at narrower angles compared to using the normal width.

**Describe the solution you'd like**
I would love to see the option for Arachne walls to add the ability to increase your wall size to help with steep sections of a print. This way the outer wall doesn't need to always be wider, just when it is needed.

**Describe alternatives you've considered**
Just always use a wider outer wall and lose details in some areas due to the wider wall.

**Additional context**
Add any other context or screenshots about the feature request here.

SaltWei, 2023-01-11:
> I would love to see the option for Arachne walls to add the ability to increase your wall size to help with steep sections of a print.

[image]
Thank you very much for suggestions.
Do you mean these options?

Quick-Flash, 2023-01-11:
> > I would love to see the option for Arachne walls to add the ability to increase your wall size to help with steep sections of a print.
> 
> [image] Thank you very much for suggestions. Do you mean these options?

These settings only give the ability to set how small you would like the walls to be. I'd like options where the wall detects overhang, rather than how thin the wall is, and has a Maximum wall width for overhangs. Currently I can sort of make it work by increasing the outer wall width, but I would rather not always have a wider outer wall, but only have one when it could help with overhangs. 

Here are some examples of wider extrusions helping with overhangs.
[image]
[image]
More info can be found here https://toms3d.org/2022/09/22/get-the-benefits-of-a-0-6mm-nozzle-with-a-0-4/