bambulab/BambuStudio #1044: "First Layer" Nozzle Temperature is not being used (with gcode example)

AddChickpeas, 2023-01-10:
**Describe the bug**
When a "First Layer" temperature is set in Filament Settings, the first layer is not printed using this temperature. 

I'm trying to print the first layer at 225 with the rest at 215. 

It seems to briefly set the temperature to 225 after homing, but then before it draws the nozzle load line it drops it to 215. 

In the attached 3mf, at line 503 it sets the temperature to 225: 

`M104 S225 ; set extrude temp earlier, to reduce wait time`

But then it overrides this setting in the nozzle load line section at line 524: 

```
;===== noozle load line ===============================
M975 S1
G90 
M83
T1000
G1 X18.0 Y1.0 Z0.8 F18000;Move to start position
M109 S215
G1 Z0.2
G0 E2 F300
G0 X240 E15 F6033.27 
G0 Y11 E0.700 F1508.32
G0 X239.5
G0 E0.2
G0 Y1.5 E0.700
G0 X18 E15 F6033.27 
M400
```

**3mf File for This Bug**

[wallet.gcode.zip](https://github.com/bambulab/BambuStudio/files/10378475/wallet.gcode.zip)

If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**

1. Create a new filament profile 
2. Set "First Layer" Nozzle Temperature to 225
3. Set "Other Layers" to 215
4. Print any file 

**Expected behavior**
The entire first layer is printed at 225. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 
 - Version 1.4.0.18

MadMax389, 2023-01-10:
Silly question, but how are you able to examine the gcode inside the exported .3mf file?

b-reich, 2023-01-10:
> Silly question, but how are you able to examine the gcode inside the exported .3mf file?

you can export the gcode

MadMax389, 2023-01-10:
@b-reich Yes but that creates a .3mf file that I am unable to read with Notepad++. However I do see an "export gcode" option under the file menu.
[image]

Hologos, 2023-01-10:
> @b-reich Yes but that creates a .3mf file that I am unable to read with Notepad++.

`.3mf` files are just zip files. If you rename it to `.zip` and unzip it, you get a folder structure where you can find the gcode. ;)

AddChickpeas, 2023-01-10:
I just used 7zip to unzip the 3mf. No file name change required.

MadMax389, 2023-01-10:
Thanks. Didn't realize you could extract the .3mf.

SaltWei, 2023-01-11:
> But then it overrides this setting in the nozzle load line section at line 524:

The start Gcode of P1P is wrong.
It shoule be "M109 S{nozzle_temperature_initial_layer[initial_extruder]}" at this line, rather than "M109 S{nozzle_temperature[initial_extruder]}".

P1P is wrong, and X1 is right.

We will fix this error. Thank you very much.

AddChickpeas, 2023-01-11:
Awesome. Just updated my start Gcode. Thanks for the quick response.

SaltWei, 2023-01-28:
Has been fixed in 1.4.2 version

closed 2023-01-28 by SaltWei