bambulab/BambuStudio #1049: Add Cura "snap to build plate" toggle option

Mypd1991, 2023-01-10:
**Is your feature request related to a problem? Please describe.**
Currently users cannot move an object above or below the build plate. Without some workarounds.

**Describe the solution you'd like**
Add the Cura toggle option of "snap to build plate".
This would allow users to move an object above or below the build plate to be fully supported by "supports" or be partially cut without have to use the cut tool. 
This would also negate having to add an additional superfluous object attached to another object to be able to move it off the build plate. This is the workaround currently having to be used to place an object above the build surface.

**Describe alternatives you've considered**
Workaround noted above

SaltWei, 2023-01-11:
https://github.com/bambulab/BambuStudio/issues/973

Bangwa88, 2023-01-31:
Pls also add slincing tolerance = exclusive from cura. We Need it for Threads and screws

icerabbitje, 2023-02-26:
This is absolutely needed ... along with serious improvements in doing assemblies on the plate in multiple colors.
I can't believe what a pain it is to try to line up multiple individual parts in different colors.