bambulab/BambuStudio #1050: [Feature request] Bigger Checkboxes on the touchscreen

NairoDorian, 2023-01-10:
**Is your feature request related to a problem? Please describe.**
Yes I and many others  are having issues with the checkbox on the touchscreen of the X1/X1C at the start of a print, to click on the checkbox, they are sooo tiny and induce missclick, I am talking about the Bed Leveling, Calibration and TImelapse checkboxes

**Describe the solution you'd like**
There is som much free space, please make the checkboxes at least twice at big to allow easy checking and unckeck

I am not the first to report it, I have seen many videos like this before but no one took the time to post a feature request, so here it is.

https://www.facebook.com/groups/bambulabofficialusers/posts/933627724247914

SaltWei, 2023-01-11:
This has no relationship with Slicer and maybe official support can't be got timely here.
I have transfered this ticket to printer team.

For firmware and printer issue, please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better support. Thanks!

sodachen22, 2023-01-11:
@NairoDorian Regarding the small checkboxes, it is already changed in the latest firmware that we make both the checkboxes and the text of the checkbox clickable.
But there is still a delay between the user clicking the checkbox and the checkbox changing states. That is because UI is waiting for the status to change from execution MCU after checking box is clicked.