bambulab/BambuStudio #1051: [FR] Mesh Leveling Taper

CCS86, 2023-01-11:
Correct me if I am wrong, but mesh leveling seems to be applied globally, throughout the full height of the print.

IMO having a configurable (or even fixed) height for which the mesh corrections are "tapered" out, is much more preferable. It's obvious that a mesh Z correction is necessary to get a consistent first layer. But especially given the print speeds of the X1C, this correction should blend to zero as quickly as possible. I believe that the Z axis struggles to keep up with the relatively large mesh corrections being made and that desynchronization between these Z corrections and the nozzle position causes some extrusion inaccuracy on lower layers.

Also, printing a non-flat top surface gives no benefit and a flat surface would always be preferred.

One approach could be to allow a maximum mesh correction change of say 0.02mm per 1mm print height. Then, using the maximum mesh correction needed, you calculate a height where the correction will be zero and up to that height, the mesh correction values are scaled incrementally.

Thanks for considering!

SaltWei, 2023-01-11:
> One approach could be to allow a maximum mesh correction change of say 0.02mm per 1mm print height. Then, using the maximum mesh correction needed, you calculate a height where the correction will be zero and up to that height, the mesh correction values are scaled incrementally.

Thank you very much for suggestions!
But this seams to be handled by printer and maybe official support can't be got timely here(For slicer).

I have transfered this ticket to printer team.

For future issue about firmware and printer, please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better support. Thanks!

NairoDorian, 2023-01-19:
CCS86 Did you follow up ?

SaltWei, 2023-01-21:
Response from firmware team： We have tried this , but the test result shows that there was no print quality enhancement , but introduced dimension deviation . So we did not implement this solution.

CCS86, 2023-01-21:
I would argue that tapering off the mesh compensation just changes the type of dimensional deviation, it doesn't introduce it. It's the lack of flatness of the bed that introduces the deviation. For obvious reasons, the nozzle must track the bed deviation to begin with. 

If you continue to apply full mesh compensation, the top of the print will be parallel to the bottom. So, if you measure local heights they will be close to the same. But, the top has the same flatness deviation as the bottom now, so it is as if the whole print is warped.

What magnitude of deviation is typical from the mesh leveling? Is there some way we can output the results from bed meshing? 

Besides the lack of flatness, the quality issue I am seeing here is that the mesh compensation causes local extrusion error. My theory is that because the printer is so fast, and has only a single Z motor driving 3 lead screws via belt, the bed cannot keep up with the required acceleration demanded by the print head speed and amount of mesh compensation to be applied. Since the first layer is slow, this is not an issue. If the mesh compensation then tapered off, over some number of layers, it would be less of an issue. But full mesh compensation for the entire print ensures that any local extrusion error persists for the entire print. It also causes extra wear and tear on the Z axis parts.

NairoDorian, 2023-01-21:
Being able to see the results of the bed leveling, with a GUI much like octoprint would be the perfection, so we have an idea of the intensity of deviation and the locations as well so w might be able to tram more accurately the bed

CCS86, 2024-04-03:
@sodachen22 any updates on this?

CCS86, 2024-04-03:
> > hello CCS86，
> > Here communicated with our R&D team.
> > Actually we tried this in this way last year to improve print surface quality.
> > But we found that the printing quality of Z axis will be lost. That is why we failed this method.
> > So do you mind the lost quality of Z-axis?? Maybe for some filaments, it is not very obviously than the top surfaces.
> > What is your opinion??

Do you have any example photos showing the quality loss you are reporting? 

Thousands of people have used this functionality on Duet3D controlled printers for years, and I have never seen a reported issue. 

Let's say the mesh leveling max correction is 0.3mm (that's pretty large), and we taper the mesh compensation over 10mm (50 layers @ 0.2mm), that would be only a 0.006mm deviation per layer. Obviously, you can increase the tapering difference to minimize this further.