bambulab/BambuStudio #1052: Manual temperature change resets on filament changes

TravisL217, 2023-01-11:
**Describe the bug**
Might not be a bug necessarily.  When sending a print job to the printer sometimes I'll manually change to sport mode and increase the temperature from the X1's screen or in the Handy App.  I noticed when it did color changes it reset the temperature back down to whatever the slicer had in the g-code I assume, but not the speed.  This causes issues as the filament can't keep up with the speed since it's now a lower temp.

**To Reproduce**
Steps to reproduce the behavior:
1. Slice a multi-color file.
2. Once printing, set temperature manually on the X1C screen
3. Wait for filament swap. 
4. Verify temp changes back.

**Expected behavior**
If the temp is set manually after the print has started, the temp should remain constant throughout the print.  

Optionally:  If the temperature is changed manually from the screen, calculate the difference from the g-code to the manual setting and maintain that differential throughout the print.  Example:  Printing ABS and PLA.  Slicer temperature settings are 200 for PLA and 220 for ABS.  Manually set slicer to 230 and sport speed.  PLA will print at 230 and ABS will print at 250.  

**Desktop (please complete the following information):**
 OS:  Windows 10

Bismar, 2023-12-15:
Why is this bug still not fixed?

Briones, 2024-09-03:
I was wondering the same. It happened with my A1, I noticed that one of my filaments needed another speed, I adjusted but after the filament change it resets to the one in the GCODE.  And since it has a lot of filament changes I can't do anything about it.