bambulab/BambuStudio #1055: infill/inner/outer wall_infill_order change

Gxiraudon, 2023-01-11:
I would like an improvement on the infill-wall order, this not a serious issue right now but when slicing a model with multiple holes,
the slicer does not really follow the infill/inner/outer mode, what it does is print infill, then the inner perimeter and outer perimeter of one individual hole, then proceed with the inner perimeter and outer perimeter of the next hole. what it should do is after printing the infill, it should print inner perimeters per hole first, then print all holes' outer perimeters. 

a txt file of a gcode is included to observe the behavior when printing the holes (between the spool spokes in this gcode). It prints the inner and outer perimeters per hole. (change .txt file extension to .gcode)
[0.5 kg spool p1 AMS adaptor_PLA_1h55m.txt](https://github.com/bambulab/BambuStudio/files/10393058/0.5.kg.spool.p1.AMS.adaptor_PLA_1h55m.txt)

after printing the infill, I want the slicer to print the inner perimeters of all holes before printing the outer perimeters.

ShryuKt, 2023-01-21:
To me, this is more of a bug.
In this example, the two holes in the red area are always printed with the outer perimeter first (inside/outside/fill order).
[image]

On this part, these two holes are overhanging, which makes the printing fail because the outer perimeters are printed on a void without support.
All other holes are printed in the correct order.

SaltWei, 2023-03-02:
@QingZhangBambu Please check whether this is fixed in 1.5.0