bambulab/BambuStudio #1056: Separate x axis y axis offsets

Anothermbdusted, 2023-01-11:
Would like to see separate x axis and y axis offsets instead of them being tied together as sometimes your one axis might be.off but the other is perfect.

YifanWuBambu, 2023-03-02:
@Anothermbdusted 
Do you mean the extruder x/y offset? It is seperated already.
[image]

If you mean the other one, please let me know. Thanks!

Anothermbdusted, 2023-03-02:
no i mean as for the xy offset for a print. like if a printed part is say .4 off in the x axis and .9 off in the y axis you can adjust them separately accordingly. currently it has to be together no matter what but would is it even possible to separate them?

YifanWuBambu, 2023-03-02:
@Anothermbdusted 
By offset, you mean shrinking or expanding in X/Y axis?
[image]

If it is still not the setting you mean, please help take a picture and mark it. Thanks!

Anothermbdusted, 2023-03-02:
correct but separating the x and y so you can adjust them individually instead of together. so you have a 
x hole compensation
y hole compensation

x contour compensation
y contour compensation

not x-y hole
not x-y contour

YifanWuBambu, 2023-03-02:
It is hard to do this for now. The clipper library used in the slicing process only support uniform offset.

Anothermbdusted, 2023-03-02:
It was a thought and tyvm for your time I appreciate it a bunch..

YifanWuBambu, 2023-03-02:
Thanks for your understanding : )

closed 2023-03-02 by YifanWuBambu